bluesky/ophyd #808: braces in PV names requires 4 sets of braces in prefix

JunAishima, 2020-02-25:
`#!/usr/bin/env python3

from caproto.server import pvproperty, PVGroup, ioc_arg_parser, run

class ZebraIOC(PVGroup):
    PC_DIR = pvproperty(value=1, mock_record='ai')
    PC_PULSE_STEP = pvproperty(value=10, mock_record='ai')

if __name__ == '__main__':
    ioc_options, run_options = ioc_arg_parser(
        default_prefix="C-ES:FMX{{{{Zeb:3}}}}:",
        desc='run an IOC to mock the Zebra box')
    ioc = ZebraIOC(**ioc_options)
    print('PVs:', list(ioc.pvdb))
    run(ioc.pvdb, **run_options)
`

Reducing the number of braces in the prefix C-ES:FMX{Zeb:3}: results in the following output:
`(local_lsdc) [jaishima@130-199-219-91 ophjects]$ python test2.py 
/home/jaishima/anaconda3/envs/local_lsdc/lib/python3.7/site-packages/caproto/_utils.py:246: UserWarning: EPICS_CA_ADDR_LIST is set but will be ignored because EPICS_CA_AUTO_ADDR_LIST is not set to 'no'. EPICS_CA_ADDR_LIST='127.0.0.1' EPICS_CA_AUTO_ADDR_LIST=''
  result.get('EPICS_CA_AUTO_ADDR_LIST', ''))
Traceback (most recent call last):
  File "test2.py", line 13, in <module>
    ioc = ZebraIOC(**ioc_options)
  File "/home/jaishima/anaconda3/envs/local_lsdc/lib/python3.7/site-packages/caproto/server/server.py", line 1144, in __init__
    self.prefix = expand_macros(prefix, self.macros)
  File "/home/jaishima/anaconda3/envs/local_lsdc/lib/python3.7/site-packages/caproto/server/server.py", line 975, in expand_macros
    return pv.format(**macros)
KeyError: 'Zeb'
`

danielballan, 2020-02-25:
I think you mean this to be a caproto issue.

JunAishima, 2020-02-25:
oh my...
not sure what went through my head 🙂

Jun

JunAishima, 2020-02-25:
Wrong project, this has also been correctly submitted to caproto. closing.

closed 2020-02-25 by JunAishima