bluesky/ophyd #814: semantics of Signal.value have changed

tacaswell, 2020-03-06:
In 8acc424b325e3447db2ac4601e7bc0c6d1403532 which came in via https://github.com/bluesky/ophyd/pull/653 the meaning of `sig.value` changed from "go consult the control system" to "here have my cached value".  For `Signal` this isn't a problem (as setting it will update the cache) and for `EpicsSignal` with auto-monitoring this will also be OK, but for `EpicsSignal` it means that it returns stale values (which caused confusion at a beamline which is testing the new software [1]).  A quick search across the profiles at NSLS-II shows something like ~1.2k usages of `.value` across 20 beamlines.

Short term, I think the correct thing to do is remove the cache logic from `signal.value` because this is a silent gotcha for our users.

Long term do we:
 - want leave it as a maybe-slow attribute?
 - make it raise on EpicsSignal is not monitoring?
 - always auto-monitor with EpicsSignal?
 - deprecate 

I can see a case for all of them.

[1] https://github.com/NSLS-II-SIX/profile_collection/issues/29

prjemian, 2020-03-07:
> [1] NSLS-II-SIX/profile_collection#29

Big problem for both XPCS & USAXS, this is XPCS:
https://github.com/BCDA-APS/apstools/issues/270

prjemian, 2020-03-07:
Monitor EpicsSignal is the assumption at APS, stunned to find that had not
been the case

The monitor should be the most efficient use.  It only updates on new
information.  There are a few fields which do not post monitors (consult
EPICS record reference manual) but those fields can be handled.

tacaswell, 2020-03-07:
Early on we had two concerns with "monitor and cache".  One was overwhelming the client with too many updates (particularly if you subscribe to a couple of noisey and/or big PVs that you don't actually need (like the Array pvs on AD or some of the stats values).  The other, a bit ironically, is that the protocol specifies that monitors can be dropped on the floor so we were worried about getting stale information.

----

@prjemian Is there already an ophyd issue that I missed?  This is the sort of thing that should definitely be reported upstream!

prjemian, 2020-03-07:
Thanks.  I will next time.

tacaswell, 2020-03-07:
As a rule of thumb, if something changes in a way that requires client code to change and it is _not_ documented as an intentional API change, then we want to know (as that is a bug).  If it is documented as an API change and it is from an rc or master we also want to know (as that is a chance to talk us off of that ledge ;) )