bluesky/ophyd #825: Review of the Status API

danielballan, 2020-03-27:
This issue has two goals:

1. Draft developer-oriented documentation about what an ophyd Status object is and the intent behind its design.
2. Propose possible extensions of that API to be tackled in one or more PRs.

As we build out the documentation, it may be useful to revisit these early design choices and understand them better, even if we elect not to change or extend anything. This could be also be useful for future work around handling failure (e.g. alarm states) better.

## Comparison between Status and Future

Ophyd Status objects are like [`concurrent.futures.Future`](https://docs.python.org/3/library/concurrent.futures.html#concurrent.futures.Future) objects but with some semantics specific to controlling physical hardware. How do they compare?

* Both have methods for adding callbacks that get called once upon completion (success or failure); `Status.add_callback(f)` and `Future.add_done_callback(f)`. A Status objects calls `f()` with no arguments; a Future calls `f(future)` with a reference to itself.
* Futures can be canceled until the executor begins the associated work and marks them as un-cancel-able. Status objects can never be cancelled.
* Many (not all) Status objects have a reference to the `Device` that they are moving. This enables stopping an action when it is already in progress, `status.device.stop()`, which is something a Future cannot do.
* Futures have _results_ which are set by the executor via `Future.set_result(result)` and collected by the caller via `Future.result(timeout=None)`. In the event of failure, an exception is set via `Future.set_exception(exc)`, which can be collected via `Future.exception(timeout=None)`. Calling `Future.result()` on a failed Future raises the exception. Status objects have no result to communicate back; they only have to communicate whether they are in flight, done successfully, or failed. They can log exceptions but do not provide a way to access them programmatically. They provide a boolean `success` property.
* `Status` has a `done` read-only property; `Future` has a `done()` method with the same meaning.
* Both have a concept of `timeout` but with different meanings. The `Status` object takes in `timeout` at `__init__` time and keeps its own clock starting from `__init__`. A `Future` leaves it up to the caller of `result()` or `exception()` to provide an optional `timeout`, and the clock starts when the method is called.
* Status also has a concept of `settle_time`, a constant time added after the underlying task has completed and before it reports being done.
* Status has special methods for reporting incremental progress in order to support progress bars.
* Status supports `__and__`.
* Future takes no arguments at `__init__`. Status takes in `timeout` and `settle_time`, which is needs to do in order to support its different meaning of `timeout`. It also takes in `done` and `success` at `__init__`.

## Possible Changes

1. Should we change the expected signature to callbacks from `f()` to `f(status)`? We made an analogous change in https://github.com/caproto/caproto/pull/547, and the result was nice. Crucially, it can be done in a backward-compatible way by inspecting the signature of `f` when it is registered. If we done this, the bluesky RunEngine can be simplified because it will not need to keep a separate record of the mapping between callbacks and status objects.
2. Does cancellation make sense? I think it could be done: just as with Futures, there is point past which the action of a Status cannot be canceled, but it could be canceled up until the `Write[Notify]Request` is sent to the IOC. But I think it would not be as useful in our context as it is in the general Futures context. The difference is, Futures typically spend time waiting in a queue for a resource-bound Executor, whereas Status objects have their underlying messages sent quickly. At most, we should wait until we have a compelling reason to need this.
3. Should we add `set_exception(exc)` and `exception(timeout=None)`? Seems pretty useful! We might slowly deprecate the method `_finished(success=True)` in favor of new methods `set_finished()` (akin to `set_result(result)`) and `set_exception(exc)`) so that failure always comes with an exception giving information about the cause.
4. Currently, the caller has to check `success` and `done` to know the state of the Status. (The `success` attribute is initialized to `False`, "guilty until proven innocent". This has been a point of confusion in the past.) Would it be useful to add a method called `wait(timeout=None)` either returns `None` (successfully completed), raises `WaitTimeoutError` (didn't complete within the `timeout` in the method call but still running), or raises an `exc` if an exception has been set? This would be analogous to `Future.result(timeout=None)` but without any result to return.
5. The `__and__` method is an interesting case. Why doesn't `concurrent.futures` need something like this? I think it's because we require `Device.set` to return exactly one Status object. Therefore, in order to communicate, "Wait for this and that and also that," we need to build a Status object that composes several status objects with logical AND, and return that. If we allowed `Device.set` to return a _list_ of Status objects, we could do something like `executor.gather` on the caller side, and there would be no need for `__and__`. It would also give progress bars more ability to understand what is going on. (This could be even more important for the telemetry work.) Should we extend the expected interface of `set` (and friends) to accept list of Status objects as well as just one?

I think (1), (3), and (4) may be worth doing now. I'm not convinced we ever need to do (2). I think (5) could be important but needs careful thought because it interacts with may possible future plans and needs.

danielballan, 2020-03-27:
Food for thought @MikeHart85 @klauer @hhslepicka @ZLLentz @tacaswell and anyone else interested....I have bounced these ideas off of anyone yet, just got thinking about the Futures API this morning. :- )

tacaswell, 2020-03-28:
1. :+1: to changing the signature, would make state management much easier
2. :-1: to adding a 'cancel' to the status object, cancellation is better handeled else where on the devices (who then mark the sattus objects as failed) 
3. :+1: to adding `set_exception` and 75/25 on moving to `set_finished`
4. +.75 on the wait.  We talked about that and rejected it early on (there is `wait` helper function though iirc?) .  My vague memory is that it was something about blocking semantics, but I don't recall clearly and we understand all of this stuff much better now so worth re-considering
5. There are external functions for "wait for all of these futures" and "wait for the first of these futures" that do AND and OR.  I am skeptical of expanding the return of set to be "one or a list" as the type stability seems a high cost to pay for slightly better progress bars (I suspect we could get the improvement on progress bars by tweaking that API to always be a list-like rather than scaler like).

ZLLentz, 2020-04-09:
More than once I've wished we had something like numbers 3 and 4 in that possible changes list.

danielballan, 2020-08-12:
Closed by #836 and #837

closed 2020-08-12 by danielballan