bluesky/ophyd #834: Call to `clear_auto_monitor()` in `release_pvs()` results in exception on program exit

CasperVector, 2020-04-08:
As the title, the following script works with ophyd v1.3.3:
```python
from ophyd import AreaDetector, SingleTrigger
class MyDetector(SingleTrigger, AreaDetector): pass
det = MyDetector("13SIM1:", name = "det")
```
However, with v1.4.0, the following exception is thrown on exit:
```
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 346, in get_provider
KeyError: 'epics.clibs'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/epics/ca.py", line 373, in initialize_libca
  File "/usr/local/lib/python3.6/site-packages/epics/ca.py", line 330, in find_libca
  File "/usr/local/lib/python3.6/site-packages/epics/ca.py", line 280, in _find_lib
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 1132, in resource_filename
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 348, in get_provider
ImportError: sys.meta_path is None, Python is likely shutting down
```

With `git bisect`, the bug came into existence at commit a0c602ad, and the problem can be worked around with the following patch:
```diff
--- a/ophyd/_pyepics_shim.py
+++ b/ophyd/_pyepics_shim.py
@@ -119,7 +119,6 @@ def release_pvs(*pvs):
         pv._reference_count -= 1
         if pv._reference_count == 0:
             pv.clear_callbacks()
-            pv.clear_auto_monitor()
             if pv.chid is not None:
                 # Clear the channel on the CA-level
                 epics.ca.clear_channel(pv.chid)
```
But I do not know whether this is the correct fix.

danielballan, 2020-04-08:
Thanks for the clear report @CasperVector. Would you also tell us which version of pyepics you are using?

CasperVector, 2020-04-08:
I am using pyepics 3.4.1 with EPICS base 7.0.3.

MikeHart85, 2020-04-09:
Haven't been able to reproduce this so far with a variety of versions, for example...

```
pyepics                   3.4.1
epics-base                3.15.6
bluesky                   1.6.0
ophyd                     1.4.1
```
... seems to work fine, at least with a caproto spoofed beamline, to simulate the IOC.

One thing I haven't tried yet is EPICS base 7.0 instead of 3.15, will try that next.

danielballan, 2020-04-09:
This is mostly unhelpful, but I can confirm having seen that traceback in CI output at some point recently while focused on something totally unrelated (might have been in the databroker CI). So I can offer, if not an existence _proof_ an existence _rumor_ that this is reproducible in CI.

klauer, 2020-04-09:
pyepics issue with the same text: https://github.com/pyepics/pyepics/issues/89

CasperVector, 2020-04-10:
I can reproduce this with the following packages in a CentOS 7.7.1908 virtual machine (the "GNOME Desktop" profile with "Development Tools" enabled during installation): EPICS 3.15.6, pyepics 3.4.1, ophyd 1.4.1, numpy 1.18.2, networkx 2.4, decorator 4.4.2.

danielballan, 2020-04-13:
I just got this locally while running sphinx. Anything I can share about my environment to help, @MikeHart85?

```
Traceback (most recent call last):
  File "/home/dallan/.conda/envs/bluesky-tutorial-sphinxified/lib/python3.7/site-packages/pkg_resources/__init__.py", line 360, in get_provider
    module = sys.modules[moduleOrReq]
KeyError: 'epics.clibs'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dallan/.local/lib/python3.7/site-packages/epics/ca.py", line 373, in initialize_libca
    libca = load_dll(find_libca())
  File "/home/dallan/.local/lib/python3.7/site-packages/epics/ca.py", line 330, in find_libca
    return _find_lib('ca')
  File "/home/dallan/.local/lib/python3.7/site-packages/epics/ca.py", line 280, in _find_lib
    dllpath = resource_filename('epics.clibs', clib_search_path(inp_lib_name))
  File "/home/dallan/.conda/envs/bluesky-tutorial-sphinxified/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1145, in resource_filename
    return get_provider(package_or_requirement).get_resource_filename(
  File "/home/dallan/.conda/envs/bluesky-tutorial-sphinxified/lib/python3.7/site-packages/pkg_resources/__init__.py", line 362, in get_provider
    __import__(moduleOrReq)
ImportError: sys.meta_path is None, Python is likely shutting down
Traceback (most recent call last):
  File "/home/dallan/.conda/envs/bluesky-tutorial-sphinxified/lib/python3.7/site-packages/pkg_resources/__init__.py", line 360, in get_provider
    module = sys.modules[moduleOrReq]
KeyError: 'epics.clibs'

During handling of the above exception, another exception occurred:
```

MikeHart85, 2020-04-13:
Is it reproducible / happening consistently?

What exact python version did you run that with?

Between you getting the same issue in a very different context, and reading through the pyepics issue that @klauer linked to, it looks to me like this isn't really a problem in ophyd or bluesky as with pyepics cleanup on shutdown. One they thought they had fixed.

What I can't figure out is why it is showing up again now, and what specific circumstances trigger it.

danielballan, 2020-04-14:
Hm, sounds like a pyepics issue then. Still might be worth pursuing and submitting an upstream patch.

It is reproducible for me. Here is minimal example.

I am running a caproto IOC just to give ophyd something to connect to. I don't think the details of which PV or the fact that it's a caproto-provided PV are pertinent.

```sh
python -m caproto.ioc_examples.random_walk
`
```

And here is a how I reproduce the error:

```sh
$ python -c "import ophyd; ophyd.EpicsSignal('random_walk:x')"
Traceback (most recent call last):
  File "/home/dallan/.conda/envs/bluesky-tutorial-sphinxified/lib/python3.7/site-packages/pkg_resources/__init__.py", line 360, in get_provider
KeyError: 'epics.clibs'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dallan/.local/lib/python3.7/site-packages/epics/ca.py", line 373, in initialize_libca
  File "/home/dallan/.local/lib/python3.7/site-packages/epics/ca.py", line 330, in find_libca
  File "/home/dallan/.local/lib/python3.7/site-packages/epics/ca.py", line 280, in _find_lib
  File "/home/dallan/.conda/envs/bluesky-tutorial-sphinxified/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1145, in resource_filename
  File "/home/dallan/.conda/envs/bluesky-tutorial-sphinxified/lib/python3.7/site-packages/pkg_resources/__init__.py", line 362, in get_provider
ImportError: sys.meta_path is None, Python is likely shutting down
$
```

Version info:

```
$ python -VV
Python 3.7.7 (default, Mar 26 2020, 15:48:22) 
[GCC 7.3.0]
$ python -c "import epics; print(epics.__version__)"
3.4.1
$ python -c "import ophyd; print(ophyd.__version__)"
1.4.1
```

CasperVector, 2020-04-14:
Thanks to the hint by @danielballan, here is a simpler way to reproduce the issue.
* Obtain the following files: `CentOS-7-x86_64-Everything-1908.iso` from a CentOS mirror; `base-3.15.6.tar.gz` from the EPICS homepage; `decorator-4.4.2-py2.py3-none-any.whl`, `networkx-2.4-py3-none-any.whl`, `numpy-1.18.2-cp36-cp36m-manylinux1_x86_64.whl`, `ophyd-1.4.0-py3-none-any.whl`, `pyepics-3.4.1.tar.gz` from a PyPI mirror.
* In a QEMU virtual machine with freshly created 16GB virtual HDD, install CentOS with the `.iso` file above using the "GNOME Desktop" profile with "Development Tools" enabled.
*  Mount the `.iso` above in the virtual machine, and set the CentOS installation repository to the mount point; put the files above in a freshly created temporary directory, and create the following files in the same directory:
```
# test.db
record(ao, "test:det")
{
	field(VAL, "0")
}

# profile.sh
EPICS_BASE=/epics/base
export EPICS_BASE
EPICS_HOST_ARCH=linux-x86_64
export EPICS_HOST_ARCH
PATH="$PATH":"$EPICS_BASE"/bin/"$EPICS_HOST_ARCH"
export PATH
```
* In the same temporary directory, run the following script as root:
```
yum -y install readline-devel python3
mkdir /epics
tar -C /epics -xf base-*.tar.gz && mv /epics/base-* /epics/base
cd /epics/base && make -j "$(nproc)" && cd -
pip3 install pyepics-*.tar.gz *.whl
```
* In the same directory, run `. ./profile.sh && softIoc -d test.db`; in another shell and the same directory, run `. ./profile.sh && python3 -c 'import ophyd; ophyd.EpicsSignal("test:det")'`.

MikeHart85, 2020-04-15:
The error is happening during PV teardown while Python is shutting down. It is quite similar to the issue @klauer linked to, and that helped a lot in tracking it down.

Part of the reason I had trouble reproducing this is that I usually install these things via conda, including `pyepics`, from our nsls2forge channel ([here](https://anaconda.org/nsls2forge/pyepics)). 

Using our version of `pyepics 3.4.1`, I do not get the error. 
Using `pyepics 3.4.1` from PyPI, I do get the error (even with everything else installed via conda).

The main difference between those versions is that the PyPI version ships with `libca.so` and `libComPYEPICS.so` (which are compiled against Epics Base 3.16.2). Our version does not include these, and falls back on those available on the system. There's also an edit to the library discovery code in `ca.py`, presumably to facilitate this.

I didn't dig further into what about those particular versions of clibs might be causing this (yet), because @danielballan and I noticed that simply creating a PV in pyepics doesn't seem to trigger this in any case. So there must be something particular about ophyd triggering this.

In an effort to isolate what exactly is triggering it, I cut down `EpicsSignal` to this bare minimum:

```python
from ophyd.signal import Signal

class MinimalEpicsSignal(Signal):
    def __init__(self, read_pv):
        name = read_pv
        metadata = {}
        super().__init__(name=name, metadata=metadata)

        self._read_pv = self.cl.get_pv(read_pv)
        self._read_pv._reference_count += 1

    def destroy(self):
        '''Disconnect the EpicsSignal from the underlying PV instance'''
        super().destroy()
        if self._read_pv is not None:
            self.cl.release_pvs(self._read_pv)
            self._read_pv = None

pv = MinimalEpicsSignal('random_walk:x')
```
The actual trigger for this problem occurs inside `release_pvs`, in `_pyepics_shim.py`:

```python
def release_pvs(*pvs):
    for pv in pvs:
        pv._reference_count -= 1
        if pv._reference_count == 0:
            pv.clear_callbacks()
            pv.clear_auto_monitor()
            if pv.chid is not None:
                # Clear the channel on the CA-level
                epics.ca.clear_channel(pv.chid)

            pv.chid = None
            pv.context = None

            # Ensure we don't get this same PV back again
            epics.pv._PVcache_.pop(pv._cache_key, None)
```
Specifically, if you comment out this line...
```python
pv.clear_auto_monitor()
```
... the error stops.

`clear_auto_monitor` looks fairly innocent:
```python
    @_ensure_context
    def clear_auto_monitor(self):
        """turn off auto-monitoring"""
        self.auto_monitor = False
```
But `self.auto_monitor` is a property, and there's that decorator. Both do quite a bit, including invoking `ca`, which is already gone, because Python is closing and tearing everything down.

Don't yet know what the best solution here is. Removing that call from `release_pvs` seems to fix it, but I'm not sure if that would have unwanted side-effects. There's also the temptation to take this up with pyepics. But I'm not entirely sure whether ophyd is at fault for calling `clear_auto_monitor` during teardown, or whether pyepics is at fault for letting it fail during teardown.

Suggestions very welcome.

MikeHart85, 2020-04-15:
Might have found a reasonable solution, though it requires reintroducing threading in pyepics_shim.

@CasperVector, could you check to see if that branch works for you?

klauer, 2020-04-15:
@MikeHart85 - would you mind giving https://github.com/pyepics/pyepics/compare/master...klauer:fix_shutdown_perhaps?expand=1 a try? It patches pyepics to not reinitialize libca after finalize_libca is called.

MikeHart85, 2020-04-15:
That works!

And flipping the comment on `initialize_libca()` vs the `raise` line toggles the error on and off as expected.

A side-effect is that [this condition](https://github.com/pyepics/pyepics/blob/4b204cdf01dd1e4552848ab52e15ee165ce478bf/epics/pv.py#L1119-L1120) (introduced in https://github.com/pyepics/pyepics/pull/102) no longer works:

```python
    def __del__(self):
        if getattr(ca, 'libca', None) is None:
            return
    # ...
```

However, I think your fix also makes that check obsolete... so that's probably fine and it could be removed.

klauer, 2020-04-15:
OK, cool. Hmm, well - I think it's at least a start for a pyepics fix. Feel free to build on it as you see fit and shoot it off for review.

danielballan, 2020-04-15:
Drive-by comment without taking the time to fully understand the issue: maybe this is a job for a [finalizer](https://docs.python.org/3/library/weakref.html#comparing-finalizers-with-del-methods) instead of `__del__`.

CasperVector, 2020-04-16:
The fix by @klauer works for me.  Thanks!

MikeHart85, 2020-04-17:
I ran into some additional issues while preparing @klauer's fix for a pyepics PR and looking for better ways to test for this. Unfortunately that fix doesn't work as well as we thought.

Using a minimal script to provoke the reported issue, needing only `pyepics==3.4.1`:
```python
import epics

class Broken:
    def __init__(self, pv):
        self.pv = epics.get_pv(pv)

    def __del__(self):
        self.pv.clear_auto_monitor()

mypv = Broken('sim:mtr1')
```

Output **with** the fix suggested in https://github.com/pyepics/pyepics/commit/1bb4a68c81d5fe128cfdd43aa9cebf5d4bdeb9a2 (without, you get the reported issue):
```
$ python broken.py 
Exception ignored in: <function Broken.__del__ at 0x7f93902b5c20>
Traceback (most recent call last):
  File "broken.py", line 9, in __del__
  File "pyepics/epics/pv.py", line 55, in wrapped
  File "pyepics/epics/pv.py", line 444, in clear_auto_monitor
  File "pyepics/epics/pv.py", line 48, in wrapped
  File "pyepics/epics/pv.py", line 409, in auto_monitor
  File "pyepics/epics/pv.py", line 48, in wrapped
  File "pyepics/epics/pv.py", line 357, in _check_auto_monitor
  File "pyepics/epics/pv.py", line 962, in count
  File "pyepics/epics/pv.py", line 911, in _getarg
  File "pyepics/epics/pv.py", line 484, in get
  File "pyepics/epics/pv.py", line 48, in wrapped
  File "pyepics/epics/pv.py", line 523, in get_with_metadata
  File "pyepics/epics/pv.py", line 48, in wrapped
  File "pyepics/epics/pv.py", line 422, in wait_for_connection
  File "pyepics/epics/pv.py", line 48, in wrapped
  File "pyepics/epics/pv.py", line 437, in connect
  File "pyepics/epics/ca.py", line 574, in wrapper
epics.ca.ChannelAccessException: Unexpected channel ID
```

We weren't seeing this in ophyd since this exception was being caught and swallowed:
```python
import epics

class Broken:
    def __init__(self, pv):
        self.pv = epics.get_pv(pv)

    def __del__(self):
        try:
            self.pv.clear_auto_monitor()
            print('yay!')
        except:
            print('oops')
```

Output **with** above mentioned fix on pyepics:
```
$ python broken.py
oops
```

So next I tried a fix based on @danielballan's drive-by comment:

```python
import epics
import weakref

class LessBroken:
    def __init__(self, pv):
        self.pv = epics.get_pv(pv)
        self._finalizer = weakref.finalize(self, self.finalize)

    def finalize(self):
        try:
            self.pv.clear_auto_monitor()
            print('yay!')
        except:
            print('oops')

mypv = LessBroken('sim:mtr1')
```

Output with **and without** above fix:
```
$ python broken.py
yay!
```

TL;DR: 

Looks like @danielballan's drive-by comment was right on the money. I think that's the best way to go. I'll make an ophyd PR with this change in a bit.

MikeHart85, 2020-04-17:
I'm a bit horrified by what a nightmarish headache trap `__del__` is, and equally impressed by how slick `weakref.finalize` works. Something to remember. Thanks again @danielballan.

MikeHart85, 2020-04-17:
@CasperVector, could you please check if this fix works for you as well?

CasperVector, 2020-04-18:
Here the `Broken` class errors out with `RuntimeError: libca shutting down` (perhaps because there is actually not a `sim:mtr1` PV on the channel), and `LessBroken` works as expected.

PR #845 (2020-04-20): Use finalize instead of __del__ to tear down Signals
Fixes #834

Replaces #842

`__del__` turns out to be unreliable and causes issues when interacting with specific versions of pyepics, epics-base and libca during teardown. This is because the order that things are destroyed in is inconsistent and implementation specific.

`weakref.finalize` is used instead to avoid this problem. Thanks @danielballan for suggesting it!
closed 2020-04-20 by klauer