bluesky/ophyd #839: Deprecate setting 'done' and 'success' in StatusBase.__init__.

danielballan, 2020-04-15:
Follow-up to #837 

Should we continue to support making Status object "done" from the very start, as in `StatusBase(success=True, done=True)` or `StatusBase(success=False, done=True)`? I can't think of a reason for needing to set this at `__init__` time. A `concurrent.futures.Future` object doesn't support that, as noted in #825. Instead, it would be just as good to do:

```py
st = StatusBase(); st.set_finished()  # success=True
```

or

```py
st = StatusBase(): st.set_exception(SomeException(...))  # success=False
```

* One can set a specific reason for failure, rather than `success=False`, which sets a blunt `UnknownStatusFailure` exception lacking any information.
* One avoids the possibility of illegal combinations like `StatusBase(success=True, done=False)` ("How can you know you succeeded if you aren't done?").
* The action is clearer (subjectively).

OTOH, if we think this is widely used we can continue to support it indefinitely and just try to advise people not to use it via documentation.

ZLLentz, 2020-04-15:
Supporting these args indefinitely on `BaseStatus` does feel harmless, but like you say these should really never be set at `__init__`.

Having status objects that are immediately done is reasonably useful in the context of simulated devices, which is why `ophyd.sim.NullStatus` was created. If there is some desire for this feature outside of simulated devices, perhaps we can lift `NullStatus` out of the sim module? At least then the user intent is very clear and there is no `success=True, done=False` trap for things like `DeviceStatus`. `NullStatus` would continue to be the "don't ask me about this, it always succeeds" shortcut. But there isn't much reason to do this unless there is some clamoring for such a shortcut.

danielballan, 2020-04-15:
Yes. And of course `NullStatus` could be re-implemented as

```py
class NullStatus(StatusBase):
    def __init__(self):
        super().__init__()
        self.set_finished()
```

PR #843 (2020-04-17): Deprecate Status(..., done=True)
Closes #839

These are harmless, but I think worth deprecating and removing in the name
of simplicity.
closed 2020-04-17 by danielballan