bluesky/ophyd #840: Consider adding alternate FormattedComponent using string.Template

klauer, 2020-04-15:
This would more closely mimic how epics-base does string interpolation.
https://docs.python.org/3/library/string.html#template-strings

Copying in the documentation relevant portions and example:

   * `$identifier names a substitution placeholder matching a mapping key of "identifier". By default, "identifier" is restricted to any case-insensitive ASCII alphanumeric string (including underscores) that starts with an underscore or ASCII letter. The first non-identifier character after the $ character terminates this placeholder specification.`

   * `${identifier} is equivalent to $identifier. It is required when valid identifier characters follow the placeholder but are not part of the placeholder, such as "${noun}ification".`

   * `$$ is an escape; it is replaced with a single $.`

```python
>>> from string import Template
>>> s = Template('$who likes $what')
>>> s.substitute(who='tim', what='kung pao')
'tim likes kung pao'
>>> d = dict(who='tim')
>>> Template('Give $who $100').substitute(d)
Traceback (most recent call last):
...
ValueError: Invalid placeholder in string: line 1, col 11
>>> Template('$who likes $what').substitute(d)
Traceback (most recent call last):
...
KeyError: 'what'
>>> Template('$who likes $what').safe_substitute(d)
'tim likes $what'
```

(Had I known it existed in 2015, I would've used it in the first place...)

danielballan, 2020-04-16:
Should we call it `TemplatedComponent`? Easy enough to explain that `FormattedComponent` uses `str.format` and `TemplatedComponent` uses `string.Template`.

klauer, 2020-04-16:
`TemplatedComponent` would be the Python name
`MacroComponent` would be the EPICS name

I'll leave it at that 🤷‍♂️

danielballan, 2020-04-16:
How about `TemproComponent`? `MacplateComponent`?

danielballan, 2020-04-16:
^ PSA: This is a joke. :-D

mrakitin, 2020-04-16:
I thought it's a pair of names from the latest Star Wars 😄

prjemian, 2020-04-16:
A musical term, as in:  senza sordino TemproComponent ma non troppo

closed 2023-11-09 by klauer