bluesky/ophyd #854: Warning if Component is passed a non-class?

ZLLentz, 2020-04-28:
There was recently an issue downstream where, rather than passing a `Device` subclass into `Component`, a function that creates a device was passed in instead. This actually works ok... sometimes. It definitely breaks any code that needs to inspect the parent class's subdevices.

This isn't permitted in the `Component` docstring which specifies that it requires a `class`. Should we be throwing an appropriate warning here? E.g. if:
- we see a non-class callable
- we see something that doesn't subclass `OphydObject`
- some other criteria?

danielballan, 2020-04-28:
I am in favor of doing cheap validation in `Component.__init__`. I am against strict type-checking because I think it is a feature that we can operate with other implementations that don't necessarily import ophyd or subclass `OphydObject` as long they present the required interface (e.g. ophyd-tango).

In a conversation with @RussBerg an idea was proposed to add something like a `does_this_satisfy_the_required_device_api` function. Could we run that on the first argument to `Component`?

klauer, 2020-04-28:
If `ophyd` isn't imported, any validation done in `Component` doesn't matter, but agreed that we should not check if it's an `OphydObj` subclass.

Generic validation as you suggest, @danielballan, would fail for a callable in place of a class (e.g., `cpt = Cpt(lambda: None)`, as you couldn't inspect the device API compatibility without first calling it to inspect the type. I think that's really at the core of this issue that @ZLLentz is bringing up.