bluesky/ophyd #862: Add `DeviceKindView` - an "overlay" of sorts for alternate `kind` settings of a device

klauer, 2020-05-06:
Problem(s)
------------
* As of ophyd v1.5.0, there is no simple way to determine the actual `kind` of any given component without instantiating the entire device (that is, `MyDevice.cpt.kind` and `MyDevice().cpt.kind` may differ)
    - My issue revolves around this (as it's problematic for general device introspection, and specifically automated GUI creation via [typhos](https://github.com/pcdshub/typhos/))
    - Alternative suggestions are welcome
* There are _many_ ways to specify `kind` settings for individual signals:
    1. In the component of a device: `cpt = Component(Signal, kind='hinted')`
    2. In a keyword argument `read_attrs`/`configuration_attrs` to `Device.__init__` (currently deprecated but not yet removed)
    3. In a `Device` class body in the form of a list: `_default_read_attrs`, `_default_configuration_attrs`. These are the defaults for (2) if not specified.
    4. Various ways external to the class definition/instantiation (`device.cpt.kind = ...`)
* An [awkward (at best) class](https://github.com/bluesky/ophyd/blob/8bdd66baf7d9e5ecc09ea2a00698aed078567b40/ophyd/device.py#L1429) handles conversion between these attribute lists and kind settings
* Some forms of the above require component names to be strings (e.g., `read_attrs=['cpt']`), which can be prone to typos and may break autocompletion in IDEs

Code in place of wall of text
---------------------------
```python
import ophyd

class MyDevice(ophyd.Device):
    cpt = ophyd.Component(ophyd.Signal, kind=ophyd.Kind.hinted)
    _default_configuration_attrs = ['cpt']

# Can you guess what these might show?
print(MyDevice.cpt.kind)
print(MyDevice(name='test').cpt.kind)
print(MyDevice(name='test', configuration_attrs=[]).cpt.kind)
```

Idea
-----

1. Remove/deprecate all attribute lists (not a requirement, but this would reduce confusion)
2. Add a `DeviceKindView` overlay that meets the following criteria:
    - Behaves as the original `Device` (allowing for instantiation and `isinstance` checks)
    - Allows for changing of component `kind` (and sub-device component `kind`s) without affecting the original `Device`
    - Does not include the ability to access components by string
3. Moving forward, only use `Component(..., kind='...')` and `DeviceKindView` to make an alternate

What this might look like
-------------------------

```python
class MySubDevice:
    cpt = Cpt(Signal, kind='normal')

class MyDevice:
    cpt1 = Cpt(Signal, kind='normal')
    cpt2 = Cpt(Signal, kind='normal')
    sub = Cpt(MySubDevice, kind='normal')

# Adjust kinds by way of `setattr`
HintedViewDevice = DeviceKindView(MyDevice)
HintedViewDevice.cpt1.kind = 'config'
HintedViewDevice.sub.cpt.kind = 'hinted'

# config_attrs could potentially still be respected, now without strings:
HintedViewDevice = DeviceKindView(MyDevice, config_attrs=[MyDevice.cpt, MyDevice.sub.cpt])

# These would be valid:
assert issubclass(HintedViewDevice, MyDevice)
assert isinstance(HintedViewDevice(name='a'), MyDevice)
```

Similarly, the hinted devices could be used within class bodies if desirable, where the original class still remains intact:

```python
class MySubDevice:
    cpt = Cpt(Signal, kind='normal')

class MyDevice:
    cpt1 = Cpt(Signal, kind='normal')
    cpt2 = Cpt(Signal, kind='normal')

    HintedSubDevice = DeviceKindView(MySubDevice)
    HintedSubDevice.cpt.kind = 'hinted'
    sub = Cpt(HintedSubDevice, kind='normal')
```

I think this could alleviate most of the awkwardness, problems with dotted attribute name typos, and perhaps provide a better basis for addressing/fixing https://github.com/bluesky/ophyd/issues/613.

Thoughts?

danielballan, 2020-05-07:
Do you have in mind disallowing (iv) on instances, making kind immutable and locked in once a Device is instantiated? If we did that, the keys in read() and read_configuration() would be knowable from the class definition and stable throughout the instance’s lifecycle, which seems desirable.

Inevitable musings on naming: The word “overlay” that you used above seems like a good one, maybe better than “view”. Views (for example, in a SQL or numpy context) might lead the reader to expect that attempts to mutate them will be disallowed or ignored. DeviceKindOverlay might be more clear, for the cost of an additional 3 letters / two syllables.

klauer, 2020-05-07:
I think there are good arguments for not allowing kind changes for device instances - only device classes and overlays. I'd be on board with it.

And for naming, thinking a step ahead to the also inevitable shortening of the name: `DKO` looks like it [might be ok](https://acronyms.thefreedictionary.com/DKO). 😁

klauer, 2020-11-03:
I have no intention of implementing this stale idea.

If you do, reader, feel free to make your own issue/PR and reference this one with your proposed functionality.

closed 2020-11-03 by klauer