bluesky/ophyd #874: set motor limits ends up swapped

prjemian, 2020-06-12:
See: https://github.com/APS-USAXS/ipython-usaxs/issues/341

<details>
<summary>(original message posted here)</summary>
note this behavior (which should NOT happen):

```
In [16]: waxsx.set_lim(-5,5)                                                                                                                                            

In [17]: wa motor                                                                                                                                                       
motor
  Positioner                     Value       Low Limit   High Limit  Offset     
  camy                           0.301       -12.5       273.5       841.151    
  tcam                           -4.0        -70.36      39.54       -52.155    
  tens                           29.428909   -17.0       45.0        7.880409   
  waxsx                          0.001       5.0         -5.0        257.897    

  Local variable name                    Ophyd name (to be recorded as metadata)
  camy                                   camy                                  
  tcam                                   tcam                                  
  tension                                tens                                  
  waxsx                                  waxsx                                 

In [18]: waxsx.high_limit_travel.pvname                                                                                                                                 
Out[18]: '9idcLAX:m58:c0:m4.HLM'

In [19]: !caget 9idcLAX:m58:c0:m4.DIR                                                                                                                                   
9idcLAX:m58:c0:m4.DIR          Neg
```

Since high < low now, the motor cannot move and this lovely error dump distracts us (the problem is motor limits and the `NoneType` is the distraction) from the real problem:

```
In [14]: RE(move_WAXSOut())                                                                                                                                             
I Wed-11:08:19 - Moving WAXS out of beam
Run aborted
Traceback (most recent call last):
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/run_engine.py", line 1355, in _run
    msg = self._plan_stack[-1].send(resp)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1307, in __call__
    return (yield from plan)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 1162, in baseline_wrapper
    return (yield from plan_mutator(plan, insert_baseline))
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 170, in plan_mutator
    raise ex
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 803, in monitor_during_wrapper
    return (yield from plan2)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 170, in plan_mutator
    raise ex
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 170, in plan_mutator
    raise ex
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 861, in fly_during_wrapper
    return (yield from plan2)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 170, in plan_mutator
    raise ex
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 170, in plan_mutator
    raise ex
  File "/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/plans/move_instrument.py", line 69, in move_WAXSOut
    terms.WAXS.x_out.get() + terms.WAXS.x_limit_offset.get())
TypeError: 'NoneType' object is not iterable
[... 120 lines cut ...]
--> 170                     raise ex
    171         # if inserting / mutating, put new generator on the stack
    172         # and replace the current msg with the first element from the

/share1/bluesky/conda_envs/bluesky_2020_5/lib/python3.7/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    121             ret = result_stack.pop()
    122             try:
--> 123                 msg = plan_stack[-1].send(ret)
    124             except StopIteration as e:
    125                 # discard the exhausted generator

~/.ipython/profile_bluesky/startup/instrument/plans/move_instrument.py in move_WAXSOut()
     67     yield from waxsx.set_lim(
     68         waxsx.low_limit_travel.get(),
---> 69         terms.WAXS.x_out.get() + terms.WAXS.x_limit_offset.get())
     70 
     71     logger.info("Removed WAXS from beam position")

TypeError: 'NoneType' object is not iterable

```

</details>

It's a simple case setting high to high and low to low.  As-is, it does the opposite (ouch!):   https://github.com/bluesky/ophyd/blob/980154632b1cfa8c4cf02e42a93a521142aefcc0/ophyd/epics_motor.py#L350-L351

prjemian, 2020-06-12:
At the same time, it's great to add a logger message that reports if limits could not be set and perhaps why.  https://github.com/APS-USAXS/ipython-usaxs/commit/8d7b02c3e23a861291bfbfc34740f5ed9a98fe1b#diff-7e36c833bd50a819cf4cd8392606345fR52-R57

I see there is a logger defined but not used in this module.  https://github.com/bluesky/ophyd/blob/980154632b1cfa8c4cf02e42a93a521142aefcc0/ophyd/epics_motor.py#L14

Suggestions?

klauer, 2020-06-12:
`self.log`

stmudie, 2020-07-21:
Yes I ran into this swapped limit issue today.

prjemian, 2020-07-21:
Thanks for reminding.

PR #886 (2020-07-21): EpicsMotor: set hi limit to hi, lo to lo
fixes #874
closed 2020-07-21 by prjemian