bluesky/ophyd #891: EPICS motor records have a .DISA field. Check it before trying a move?

prjemian, 2020-08-17:
The EPICS motor record, [like all EPICS records](https://epics.anl.gov/EpicsDocumentation/AppDevManuals/RecordRef/Recordref-6.html), has a `.DISA` field that says if the record has been disabled (which means that it will not be processed).  When disabled, the motor will not accept attempts to change the target (`.VAL`) value.  It would be nice if that could be checked before trying to move the motor.  Then, a clear, concise message could be delivered about why the move was not allowed.

Here's an example of what happens now for a motor that has been disabled:

```
In [24]: movr detu.x 2                                                                       
CA.Client.Exception...............................................
    Warning: "Channel write request failed"
    Context: "op=1, channel=8idi:m90.VAL, type=DBR_DOUBLE, count=1, ctx="8idi:m90""
    Source File: ../oldChannelNotify.cpp line 158
    Current Time: Mon Aug 17 2020 16:03:25.038080194
..................................................................
  A 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^Ctrying a second time
Pausing...
The RunEngine invoked by magics cannot be resumed.                                           
Aborting...
CA.Client.Exception...............................................
    Warning: "Channel write request failed"
    Context: "op=1, channel=8idi:m90.STOP, type=DBR_SHORT, count=1, ctx="8idi:m90""
    Source File: ../oldChannelNotify.cpp line 158
    Current Time: Mon Aug 17 2020 16:03:42.591596768
..................................................................
Aborting: running cleanup and marking exit_status as 'abort'...
Run aborted
Traceback (most recent call last):
  File "/home/beams/8IDIUSER/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/run_engine.py", line 1341, in _run
    msg = self._plan_stack[-1].throw(
  File "/home/beams/8IDIUSER/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 299, in mvr
    return (yield from inner_mvr())
  File "/home/beams/8IDIUSER/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/utils.py", line 1066, in dec_inner
    return (yield from plan)
  File "/home/beams/8IDIUSER/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py", line 1075, in relative_set_wrapper
    return (yield from plan)
  File "/home/beams/8IDIUSER/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/bluesky/preprocessors.py", line 245, in msg_mutator
    ret = yield msg
bluesky.utils.RequestAbort
CA.Client.Exception...............................................
    Warning: "Channel write request failed"
    Context: "op=1, channel=8idi:m90.STOP, type=DBR_SHORT, count=1, ctx="8idi:m90""
    Source File: ../oldChannelNotify.cpp line 158
    Current Time: Mon Aug 17 2020 16:03:42.626892969
..................................................................
```

prjemian, 2020-08-17:
This is easier to implement for `EpicsMotor` since it is a `Device`.  Could this be implemented for `EpicsSignal` as well?

hhslepicka, 2020-08-17:
It will also need `.DISV` to compare with `.DISA` and obtain the result if the record is disabled or not. 👍