bluesky/ophyd #894: Upstream UnitConversionDerivedSignal?

danielballan, 2020-08-20:
Unit-handling keeps coming up. It seems reasonable to add a dependency on one of the unit libraries to ophyd, given ophyd's role.

Should we upstream this an accept a unit dependency?

https://github.com/pcdshub/pcdsdevices/pull/518

ZLLentz, 2020-08-20:
I think it is worthwhile to have unit handling in `ophyd`, though I'm not sure to what extent it is appropriate. Ken's unit conversion signal is a clear candidate. In the linked repository we are using `pint` as our unit dependency because it is very portable. I would understand any and all reluctance to picking up a new dependency.

prjemian, 2020-08-20:
Recently,in the coalition telco, we discussed various unit conversion packages and [pint](https://pint.readthedocs.io/en/stable/) was the preference over [cfunits](https://pythonhosted.org/cfunits/index.html) for cross-platform compatibility.

Since then, I added `pint` to the [APS standard install](https://github.com/BCDA-APS/use_bluesky/blob/da1d7109b9b3024e706c9289bf8fd639a7e52e8f/python_installation/environment_2020_9.yml) environment on the chance that someone, sometime might start using it.

This is a great move to mainstream this into regular use.  If we first provide the mechanism for handling units, we can allow the community to drive the direction as to *how* to use the units.  XPCS, for example, uses energy in `eV` while hard X-ray science techniques tend to use `keV`.

ZLLentz, 2020-08-24:
I have no strong opinions here. I simply picked the top result on a google search for "python unit conversion library", tried it, it worked for my narrow use case, and then I moved on. If `unyt` is better and similarly cross-platform I am happy to follow the collaboration.

ksunden, 2020-08-24:
Here is Nathan Goldbaum's Scipy lightning talk (incidentally just two before @danielballan) on unyt

https://youtu.be/1HDq7QoOlI4?t=1278

ksunden, 2020-09-08:
Upon looking further, I discovered Pint has [contexts](https://pint.readthedocs.io/en/0.10.1/contexts.html) which serve the main purpose I had thought was unique to Unyt. So if Pint is where the community is at, I think it works well.

danielballan, 2020-09-09:
I like the arguments that unyt makes for itself, but I'm inclined toward pint because it seems more widely used and active. GitHub finds [2053 dependent repositories for pint](https://github.com/hgrecco/pint/network/dependents) and only [44 for unyt](https://github.com/yt-project/unyt/network/dependents). The activity over the last month is greater and involves more people.

So, I move to close debate. Any concerns about pint?

danielballan, 2020-09-09:
I don't have a strong view on whether we add pint as a hard dependency or an optional one (with an `ImportError` raised if you try instantiate this object without it). It would be fine to make an optional dependency in this case, and perhaps for the best.

But based on user feedback and other conversations I foresee building stronger support for units in ophyd generally --- growing beyond capturing opaque EGU strings from EPICS --- and _eventually_ making some units library into a hard dependency, hence the careful consideration now about which library we choose.

tacaswell, 2020-09-09:
I am not clear why we need a hard dependency in ophyd rather than relying on dependency injection by making the `convert_units` function one of the parameters to `UnitConversionDerivedSignal`.

The signature of `convert_units` is `def cu(scalar_value: PlainNumber, old_unit: str, new_unit: str) -> PlainNumber` (https://github.com/pcdshub/pcdsdevices/blob/637465ed05cdfe7c1da705aa04e3ca58c5239b32/pcdsdevices/utils.py#L94-L113) I am optomistic that we can wrap every conceivable unit framework into that signature.

We may want to go all-in on pint as a community, but that seems like something that is better captured in schema/conventions in the document model rather than in a hard dependency in ophyd.

PR #978 (2021-06-01): ENH: add unit conversion derived signal from pcdsdevices
Summary
----------
* Fixes metadata keys on `FakeEpicsSignal`
* Fixes `put()` kwargs on `FakeEpicsSignal` to avoid frustratingly-common deprecation warnings
* Adds `UnitConversionDerivedSignal` with small test suite.

Closes #894 

Example
---------

This can be used as a device component like follows:

```python
    # component_name_to_convert_from = Cpt(Signal, value=0.)
    setpoint = Cpt(UnitConversionDerivedSignal,
                   derived_from='component_name_to_convert_from',
                   derived_units='s',
                [... cut]
closed 2021-06-01 by mrakitin