bluesky/ophyd #899: Revisit _ensure_trailing_slash

danielballan, 2020-08-24:
It looks like `_ensure_trailing_slash` needs some ore work

```
TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath_RBV', name='pe1c_tiff_file_path', parent='pe1c_tiff', value='Z:\\data\\pe1_data\\2020\\08\\24\\\\', timestamp=1598287423.669508, auto_monitor=False, string=True, write_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath', limits=False, put_complete=False) to value 'Z:\\data\\pe1_data\\2020\\08\\24\\/' and timed out after 10 seconds. Current value is 'Z:\\data\\pe1_data\\2020\\08\\24\\\\'.
See /home/xf28id1/.cache/bluesky/log/bluesky.log for the full traceback.
```

Full traceback:

<details>

```
In [1]: xrun(0,1)                                                                                                                                          
INFO:bluesky:Executing plan <generator object msg_mutator at 0x7fefb85bf250>
INFO:bluesky.RE.state:Change state on <xpdacq.xpdacq.CustomizedRunEngine object at 0x7fefe65a6090> from 'idle' -> 'running'
INFO: requested exposure time = 0.1 - > computed exposure time= 0.1
ERROR:bluesky:Run aborted
Traceback (most recent call last):
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 1342, in _run
    stashed_exception or resp)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/preprocessors.py", line 245, in msg_mutator
    ret = yield msg
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 1425, in _run
    new_response = await coro(msg)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 2168, in _stage
    result = obj.stage()
  File "/home/xf28id1/.ipython/profile_collection/startup/80-areadetector.py", line 284, in stage
    super().stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/base.py", line 289, in stage
    ret = super().stage(*args, **kwargs)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/device.py", line 559, in stage
    device.stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/plugins.py", line 132, in stage
    super().stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/base.py", line 289, in stage
    ret = super().stage(*args, **kwargs)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/filestore_mixins.py", line 626, in stage
    super().stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/filestore_mixins.py", line 453, in stage
    set_and_wait(self.file_path, write_path)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/utils/epics_pvs.py", line 267, in set_and_wait
    (signal, val, timeout, current_value))
TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath_RBV', name='pe1c_tiff_file_path', parent='pe1c_tiff', value='Z:\\data\\pe1_data\\2020\\08\\24\\\\', timestamp=1598287423.669508, auto_monitor=False, string=True, write_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath', limits=False, put_complete=False) to value 'Z:\\data\\pe1_data\\2020\\08\\24\\/' and timed out after 10 seconds. Current value is 'Z:\\data\\pe1_data\\2020\\08\\24\\\\'.
INFO:bluesky.RE.state:Change state on <xpdacq.xpdacq.CustomizedRunEngine object at 0x7fefe65a6090> from 'running' -> 'idle'
An exception has occurred, use '%tb verbose' to see the full traceback.
TimeoutError: Attempted to set EpicsSignalWithRBV(read_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath_RBV', name='pe1c_tiff_file_path', parent='pe1c_tiff', value='Z:\\data\\pe1_data\\2020\\08\\24\\\\', timestamp=1598287423.669508, auto_monitor=False, string=True, write_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath', limits=False, put_complete=False) to value 'Z:\\data\\pe1_data\\2020\\08\\24\\/' and timed out after 10 seconds. Current value is 'Z:\\data\\pe1_data\\2020\\08\\24\\\\'.

 

ERROR:nslsii.ipython:Attempted to set EpicsSignalWithRBV(read_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath_RBV', name='pe1c_tiff_file_path', parent='pe1c_tiff', value='Z:\\data\\pe1_data\\2020\\08\\24\\\\', timestamp=1598287423.669508, auto_monitor=False, string=True, write_pv='XF:28ID1-ES{Det:PE1}TIFF1:FilePath', limits=False, put_complete=False) to value 'Z:\\data\\pe1_data\\2020\\08\\24\\/' and timed out after 10 seconds. Current value is 'Z:\\data\\pe1_data\\2020\\08\\24\\\\'.
Traceback (most recent call last):
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3417, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-1-38895619ba44>", line 1, in <module>
    xrun(0,1)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/xpdacq/xpdacq.py", line 775, in __call__
    return super().__call__(plan, subs, **metadata_kw)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 807, in __call__
    self._resume_task(init_func=_build_task)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 931, in _resume_task
    raise exc
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 1500, in _run
    raise err
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 1342, in _run
    stashed_exception or resp)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/preprocessors.py", line 245, in msg_mutator
    ret = yield msg
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 1425, in _run
    new_response = await coro(msg)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/bluesky/run_engine.py", line 2168, in _stage
    result = obj.stage()
  File "/home/xf28id1/.ipython/profile_collection/startup/80-areadetector.py", line 284, in stage
    super().stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/base.py", line 289, in stage
    ret = super().stage(*args, **kwargs)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/device.py", line 559, in stage
    device.stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/plugins.py", line 132, in stage
    super().stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/base.py", line 289, in stage
    ret = super().stage(*args, **kwargs)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/filestore_mixins.py", line 626, in stage
    super().stage()
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/areadetector/filestore_mixins.py", line 453, in stage
    set_and_wait(self.file_path, write_path)
  File "/opt/conda_envs/collection-2020-2.0rc7-1-xpdacq/lib/python3.7/site-packages/ophyd/utils/epics_pvs.py", line 267, in set_and_wait
    (signal, val, timeout, current_value))
```

</details>

klauer, 2020-08-25:
https://github.com/bluesky/ophyd/pull/749

danielballan, 2020-08-25:
Thanks for the reminder, @klauer. Will direct DAMA effort at getting that merged now rather than enhancing this workaround.

danielballan, 2020-09-16:
Superseded by #749

danielballan, 2020-09-16:
(Which will remove this private and broken function entirely.)

closed 2020-09-16 by danielballan