bluesky/ophyd #917: EpicsSignal default timeout only applies to class it's called on

klauer, 2020-10-05:
https://github.com/bluesky/ophyd/blob/e168aea65213cd39bc5266c7b26da7a1c686463d/ophyd/signal.py#L851-L852

Implementation implies that it be run on `EpicsSignalBase`
However, users are reportedly running it on `EpicsSignal`, which means `EpicsSignalRO` retains the defaults.

prjemian, 2020-10-05:
Aha! Thanks!

danielballan, 2020-10-05:
I could argue this both ways --- that we should manage all this state in one place (`EpicsSignalBase`) and let it apply to all subclasses, or that we should let subclasses "escape" this setting and control their own destiny, so to speak. As this is an advanced feature, I lean toward making the documentation very clear on this point and seeing how far that gets us.

prjemian, 2020-10-05:
The location of the implementation is clear.  Somehow got distracted that it was on `EpicsSignalBase` but that's OK.  Best in one place only.

klauer, 2020-10-05:
An easy-to-make mistake clarified by documentation seems reasonable 👍

PR #918 (2020-10-06): Clarity how set_default_timeout interacts with inheritance.
Closes #917
closed 2020-10-06 by danielballan