bluesky/ophyd #921: A Component's doc does not propagate to the object

mrakitin, 2020-10-16:
Today I recommneded @jklynch to use the `doc="..."` kwarg to add the documentation to components of a device, but it looks like the added docstrings have no effect.

Demo:
```py
$ ipython
Python 3.7.9 (default, Aug 31 2020, 12:42:55) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.18.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from ophyd import Device, Signal, Component as Cpt

In [2]: class MyDev(Device):
   ...:     a = Cpt(Signal, doc='signal a')
   ...: 

In [3]: my_dev = MyDev(name='my_dev')

In [4]: my_dev.__doc__

In [5]: print(my_dev.a.__doc__)
A signal, which can have a read-write or read-only value.

    Parameters
    ----------
    name : string, keyword only
    value : any, optional
        The initial value
    kind : a member the Kind IntEnum (or equivalent integer), optional
        Default is Kind.normal. See Kind for options.
    parent : Device, optional
        The parent Device holding this signal
    timestamp : float, optional
        The timestamp associated with the initial value. Defaults to the
        current local time.
    tolerance : any, optional
        The absolute tolerance associated with the value
    rtolerance : any, optional
        The relative tolerance associated with the value, used in
        set_and_wait as follows

        .. math::

          |setpoint - readback| \leq (tolerance + rtolerance * |readback|)

    cl : namespace, optional
        Control Layer.  Must provide 'get_pv' and 'thread_class'
    attr_name : str, optional
        The parent Device attribute name that corresponds to this Signal

    Attributes
    ----------
    rtolerance : any, optional
        The relative tolerance associated with the value
    

In [6]: 
```

Either I don't use it correctly, or the code is not behaving as expected. Has it ever worked?

klauer, 2020-10-16:
I think I may have broken it in https://github.com/bluesky/ophyd/pull/650 or one of the Device refactors. `help(EpicsMotor)`, to my recollection, used to work.

Even without a fix, though, `doc=` is still useful if you're using the doc helpers I provided in a recent PR. Docstrings will show up in the generated Sphinx docs in a nice table.

mrakitin, 2020-10-16:
Thanks for the information, @klauer!