bluesky/ophyd #925: Why default `timeout` kwarg in EpicsSignal `.set()` not using signal's `.timeout`?

prjemian, 2020-10-27:
Why the `timeout` kwarg in EpicsSignal `.set()` method does not default to the `.timeout` value?  Looking here:  https://github.com/bluesky/ophyd/blob/3dddf111946c1eb7200361d2158445db2acfb76d/ophyd/signal.py#L1719

NOTE: The PV in use here is the `.STUP` (status update) field of an EPICS motor record.  Set it to 1, "ON" and the record support will set it back to 0 "OFF" as the motor status is updated.  Thus, the ophyd `.set` operation will not succeed since the PV will never reach the "ON" value.

This `.set()` will default the `timeout` kwarg to `None`.  Since I've set the default timeout for the signal to be 10 s, I was expecting `s.timeout` to be 10 (and it is).  

```
(bluesky_2020_9) prjemian@poof /tmp $ ipython
Python 3.8.2 (default, Mar 26 2020, 15:53:00) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from ophyd.signal import EpicsSignalBase                                                                                  

In [2]: EpicsSignalBase.set_default_timeout(timeout=10, connection_timeout=5)                                                     

In [3]: from ophyd import EpicsSignal                                                                                             

In [4]: s = EpicsSignal("sky:m1.STUP", name='s', string=True)                                                                     

In [5]: s.timeout                                                                                                                 
Out[5]: 10
```

Now, try `st = s.set("ON")` and wait 15 seconds (5 seconds beyond timeout) before printing the status.  Expecting it be done.

```
import time
st = s.set("ON")
time.sleep(15)
print(st)
```

Expecting `done`, but no.

```
In [6]: import time 
   ...: st = s.set("ON") 
   ...: time.sleep(15) 
   ...: print(st) 
   ...:                                                                                                                           
Status(obj=EpicsSignal(read_pv='sky:m1.STUP', name='s', value='OFF', timestamp=1603771932.302131, auto_monitor=False, string=True, write_pv='sky:m1.STUP', limits=False, put_complete=False), done=False, success=False)
```

---

Repeat but use a `timeout=10` kwarg with `.set()` in a new ipython session:

```
In [1]: from ophyd import EpicsSignal 
   ...: import time 
   ...: s = EpicsSignal("sky:m1.STUP", name='s', string=True) 
   ...: st = s.set("ON", timeout=10) 
   ...: time.sleep(15) 
   ...: print(st) 
   ...:                                                                                                                           
set_and_wait(value=ON, timeout=10, atol=None, rtol=None)
Status(obj=EpicsSignal(read_pv='sky:m1.STUP', name='s', value='OFF', timestamp=1603772109.341146, auto_monitor=False, string=True, write_pv='sky:m1.STUP', limits=False, put_complete=False), done=True, success=False)

```

_Originally posted by @prjemian in https://github.com/bluesky/ophyd/pull/901#issuecomment-716968738_

prjemian, 2020-10-27:
Looking at a wider block of the code: https://github.com/bluesky/ophyd/blob/3dddf111946c1eb7200361d2158445db2acfb76d/ophyd/signal.py#L1710-L1719

Suggest that the following lines be inserted before line 1714:

```python
         if timeout is None:
             timeout = self.timeout

```

There is no comparable `settle_time` on the `EpicsSignal` object, so no need to revise this kwarg, too.

prjemian, 2020-10-27:
With that local change, the test code uses the timeout applied to EpicsSignalBase:

```
(bluesky_2020_9) prjemian@poof /tmp $ ipython
Python 3.8.2 (default, Mar 26 2020, 15:53:00) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from ophyd.signal import EpicsSignalBase 
   ...: EpicsSignalBase.set_default_timeout(timeout=10, connection_timeout=5) 
   ...: from ophyd import EpicsSignal 
   ...: import time 
   ...: s = EpicsSignal("sky:m1.STUP", name='s', string=True) 
   ...: st = s.set("ON") 
   ...: time.sleep(15) 
   ...: print(st) 
   ...:                                                                                                                                            
set_and_wait(value=ON, timeout=10, atol=None, rtol=None)
Status(obj=EpicsSignal(read_pv='sky:m1.STUP', name='s', value='OFF', timestamp=1603772907.87159, auto_monitor=False, string=True, write_pv='sky:m1.STUP', limits=False, put_complete=False), done=True, success=False)

In [2]:  

```

tacaswell, 2020-10-27:
Thanks for running this to ground @prjemian !

PR #926 (2020-11-03): EpicsSignal.set() timeout should default to EpicsSignal.timeout
Changes to fix #925.  Expected the `EpicsSignal.set()` method to use the signal's `timeout` attribute.  This PR will make that happen.
commit 5d4372e4 (2020-11-03): fixes #925
closed 2020-11-03 by prjemian