bluesky/ophyd #930: UnicodeDecodeError when creating instance of EpicsSignal

prjemian, 2020-11-03:
We have some instances  of the [*scalcout*]() record, as `4idb:PRA1`, ....  When we try to create an `ophyd.EpicsSignal("4idb:PRA1")`, we get a *UnicodeDecodeError* exception because the value has a greek mu character to describe micron units in the string value.  Looks like this one (for PRA2):

[image]

<details>
<summary>full exception traceback</summary>

```
test = EpicsSignal('4idb:PRA1',name='test')

Exception occurred during callback <function PyepicsShimPV.get_all_metadata_callback.<locals>.get_metadata_thread at 0x7f8df9df6700> (pvname='4idb:PRA1')
Traceback (most recent call last):
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/ophyd/_dispatch.py", line 45, in run
    callback(*args, **kwargs)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/ophyd/_pyepics_shim.py", line 105, in get_metadata_thread
    md = self.get_all_metadata_blocking(timeout=timeout)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/ophyd/_pyepics_shim.py", line 98, in get_all_metadata_blocking
    self.get_ctrlvars(timeout=timeout)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/pv.py", line 48, in wrapped
    return func(self, *args, **kwargs)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/pv.py", line 735, in get_ctrlvars
    kwds = ca.get_ctrlvars(self.chid, timeout=timeout, warn=warn)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 637, in wrapper
    return fcn(*args, **kwds)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 1772, in get_ctrlvars
    metadata = get_with_metadata(chid, ftype=ftype, count=1, timeout=timeout,
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 637, in wrapper
    return fcn(*args, **kwds)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 1390, in get_with_metadata
    return get_complete_with_metadata(chid, count=count, ftype=ftype,
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 637, in wrapper
    return fcn(*args, **kwds)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 1538, in get_complete_with_metadata
    metadata = _unpack_metadata(ftype=ftype, dbr_value=extended_data)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 1308, in _unpack_metadata
    md['enum_strs'] = tuple(BYTES2STR(dbr_value.strs[i].value)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/ca.py", line 1308, in <genexpr>
    md['enum_strs'] = tuple(BYTES2STR(dbr_value.strs[i].value)
  File "/home/beams/POLAR/.conda/envs/bluesky_2020_9/lib/python3.8/site-packages/epics/utils3.py", line 25, in b2s
    return str(st1, EPICS_STR_ENCODING)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb5 in position 12: invalid start byte
```

</details>

We'd like the clients of the control system to not dictate the text values of these labels, or at least not raise exceptions.  (In the present state, any _bad_ value typed by a naive user could interrupt the client code.)

prjemian, 2020-11-03:
@gfabbris : Thanks for the bug report!

prjemian, 2020-11-03:
To handle this situation, the [PyEpics documentation](http://pyepics.github.io/pyepics/arrays.html?highlight=unicode#character-arrays) says:

> Note that Epics character waveforms as defined as above are really arrays of bytes. The conversion to a string assumes the ASCII character set. Unicode is not directly supported. If you are storing non-ASCII data, you would have to convert the raw array data yourself, perhaps like this (for Python3):

```
>>> arr_data = pv.get()
>>> arr_bytes = bytes(list(array_data))
>>> arr_string = str(arr_bytes, 'LATIN-1')
```

prjemian, 2020-11-03:
This code gets the same error:

```python
from epics import caget
caget('4idb:PRA1',as_string=True)
```

prjemian, 2020-11-03:
FYI: The strings are produced by this record:
[image]

prjemian, 2020-11-03:
This command `caget('4idb:PRA1',as_numpy=True)` returns the state number (4, in this case).

prjemian, 2020-11-03:
Since this is not a waveform array, how can we get at this value as a string?  The technique suggested by the PyEpics documentation [above](https://github.com/bluesky/ophyd/issues/930#issuecomment-720868088) seems out of our reach since we can't use the `as_numpy=True` kwarg with the `ophyd.EpicsSignal`.

hhslepicka, 2020-11-03:
We bumped into this issue with PyDM.

Give this a try:

`
try:
    from epics import utils3
    utils3.EPICS_STR_ENCODING = "latin-1"
except:
    pass
`

prjemian, 2020-11-03:
Thanks! That worked.

closed 2020-11-03 by prjemian

hhslepicka, 2020-11-03:
Awesome, @prjemian ! I will try to propose the fix upstream to PyEpics.

klauer, 2020-11-03:
This is the encoding caproto has used since the start. My logic was that `latin-1` is probably closest to the encoding that was expected by default in the 90s.