bluesky/ophyd #934: Is get_frames_per_point always giving 0?

danielballan, 2020-11-06:
When I run [this example script](https://gist.github.com/danielballan/3419df60330fb1543ac6306462a2e6c0#file-interactive_test-py-L31-L33) and monitor `13SIM1:HDF1:NumCapture_RBV`, it never changes from its initial value of `0`. This remains true whether the `13SIM1:HDF1:FileWriteMode_RBV` is set to `Capture` or `Stream` (as specified in `det.hdf1.stage_sigs` and verified by monitoring).

Thus, I have to override `get_frames_per_point()` to rely on `det.cam.num_images` (the _desired_ number of images) rather than `det.hdf1.num_capture`, as shown in the highlighted lines of the example script.

I notice that we override `get_frames_per_point` in the same fashion [in CSX's profile](https://github.com/NSLS-II-CSX/xf23id1_profiles/blob/4ef6a3e30b4d10c1ed33e91deb9fa36b99a60187/profile_collection/startup/csx1/devices/areadetector.py#L74-L75) and [in FXI's profile](https://github.com/NSLS-II-FXI/profile_collection/blob/5eea8486b35f45f150de501cc02d9a4690b0e626/startup/10-area-detector.py#L68-L69)

Should we make the same change upstream? Am I misunderstanding or misremembering how the original was supposed to work?

danielballan, 2020-11-06:
I notice that for the TIFF plugin we already use `num_images`:

https://github.com/bluesky/ophyd/blob/c2c46f52d8b1dcd5aa92b66891bf8eb3277c5067/ophyd/areadetector/filestore_mixins.py#L496-L507

and likewise in the HDF5 plugin designed to be used in `Single` `FileWriteMode`.

https://github.com/bluesky/ophyd/blob/c2c46f52d8b1dcd5aa92b66891bf8eb3277c5067/ophyd/areadetector/filestore_mixins.py#L467-L483

The only exception is our main HDF5 plugin, used in `Capture` or `Stream` mode.

https://github.com/bluesky/ophyd/blob/c2c46f52d8b1dcd5aa92b66891bf8eb3277c5067/ophyd/areadetector/filestore_mixins.py#L449-L464

thomascobb, 2020-11-09:
`NumCapture_RBV` seems to work correctly in our ADCore 3-9 based release:

```
tom:~ $ caput AD-SIM-01:HDF5:NumCapture 2
Old : AD-SIM-01:HDF5:NumCapture 0
New : AD-SIM-01:HDF5:NumCapture 2
tom:~ $ caget AD-SIM-01:HDF5:NumCapture_RBV
AD-SIM-01:HDF5:NumCapture_RBV 2
```

danielballan, 2020-11-09:
Thanks. I see the same behavior. Now I understand that the issue was our failure to interpret `0` as "unbounded". We need to special-case this in ophyd. Fix proposed in #934.

PR #935 (2020-11-09): Interpret num_capture=0 correctly.
As explained in the comment, the `FileStoreHDF5` class has been recording its `num_capture` signal as the number of frames that will be captures for a given acquisition. The _correct_ behavior is to interpret `num_capture=0` as a special case because that means "Capture as many as you receive." In that case, our best guess (really an upper bound) is `num_images` from the detector's camera.

Closes #934
closed 2020-11-09 by danielballan