bluesky/ophyd #964: Support EPICS scaler record .RATE field

prjemian, 2021-02-12:
The EPICS scaler record has a [`.RATE` field](https://github.com/epics-modules/scaler/blob/master/scalerApp/src/scalerRecord.dbd#L74-L79) that controls the rate at which CA monitors will be posted while the scaler is acquiring.

```
	field(RATE,DBF_FLOAT) {
		prompt("Display Rate (Hz.)")
		promptgroup(GUI_COMMON)
		special(SPC_MOD)
		initial("10")
	}
```

Make sure the scaler support in ophyd provides a signal for this field.

prjemian, 2021-02-12:
### `EpicsScaler`
https://github.com/bluesky/ophyd/blob/03585f0b02752d5975da393a889e5c5cecd06fd7/ophyd/scaler.py#L51

### `ScalerCH`
https://github.com/bluesky/ophyd/blob/03585f0b02752d5975da393a889e5c5cecd06fd7/ophyd/scaler.py#L114

prjemian, 2021-02-12:
Nothing to do except apply consistency to the `kind` attribute.  Prefer to use `Kind.omitted` since this is cosmetic while the scaler is acquiring and does not affect any data to be stored.

Opinions?

* [ ] leave it as-is
* [ ] change `EpicsScaler`
* [ ] change `ScalerCH`
* [ ] other

danielballan, 2021-02-17:
Can we imagine a situation where it might be nice to be able to lookup the intended RATE? Maybe to check it against the actual RATE or filter Runs in a search by those than have a RATE greater than some value?

I lean toward "config" because those scenarios don't sound _too_ out there to me. But perhaps those examples are too contrived and we should just move to "omitted". I would be happy either way, and I agree that consistency is the priority here.

prjemian, 2021-02-19:
>Can we imagine a situation where it might be nice to be able to lookup the intended RATE? 

Not a likely scenario.  This field is usually set _once every few years_, if that often.  That's why `omitted` v. `config`.  So, why make it into an ophyd signal at all?  One instrument asked for it.

prjemian, 2021-02-19:
This is the screen that shows the field (in the red box):
[image]

prjemian, 2021-02-26:
Proceeding with PR that makes `Kind.omitted` the consistent choice.

prjemian, 2021-02-26:
Note: attribute `auto_count_update_rate` also affected by these same comments.

PR #969 (2021-02-26): MNT #964 apply consistency to update_rate & auto_count_update_rate
For the scaler attributes `update_rate` & `auto_count_update_rate`, apply consistency to the setting of their `kind` kwarg.  Per discussion, this change will close #964, applying `omitted` to creation of these attributes.

Affects `EpicsScaler and `ScalerCH` classes.
closed 2021-02-26 by danielballan