bluesky/ophyd #966: call to wait_for_connection() in device.py needs timeout kwarg value

prjemian, 2021-02-26:
In the recent overhaul of the various timeouts (connect, read, & write), the focus was on these values for *EpicsSignalBase*.  Now, I'm seeing a connect timeout (from PyEpics) in a VM situation when connecting to the various PVs with an EPICS ADSimDetector (area detector) Device.

```python
TimeoutError: Control layer pyepics failed to send connection and access rights information within 1.0 sec
```

The problem is that the connect timeout interval is reported as 1.0 but the defaults for EpicsSignalBase have been set (previously) much larger.  

```python
# set default timeout for all EpicsSignal connections & communications
EpicsSignalBase.set_defaults(
    auto_monitor=True, timeout=60, write_timeout=60, connection_timeout=15,
)
```

The one problem location to resolve is in `ophyd.device.Component`: https://github.com/bluesky/ophyd/blob/d0bb265e7e032e73cb3a778194fe167e58c932df/ophyd/device.py#L182

Without supplying a `timeout=GET_DEFAULT_FROM_SOMEWHERE` kwarg, this call will default to `timeout=1`.

<details>
<summary>Example</summary>

```python
(base) mintadmin@mint-vm:~/.../BCDA-APS/bluesky_instrument_training$ ./blueskyStarter.sh 
Python 3.8.5 (default, Sep  4 2020, 07:30:14) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.20.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: bluesky
Activating auto-logging. Current session state plus future input saved.
Filename       : /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/.logs/ipython_console.log
Mode           : rotate
Output logging : True
Raw input log  : False
Timestamping   : True
State          : active
I Thu-20:11:10 - ############################################################ startup
I Thu-20:11:10 - logging started
I Thu-20:11:10 - logging level = 10
I Thu-20:11:10 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/collection.py
I Thu-20:11:10 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/mpl/console.py
I Thu-20:11:10 - #### Bluesky Framework ####
I Thu-20:11:10 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/framework/check_python.py
I Thu-20:11:10 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/framework/check_bluesky.py
I Thu-20:11:11 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/framework/initialize.py
I Thu-20:11:12 - using databroker catalog 'class_2021_03'
I Thu-20:11:12 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/framework/metadata.py
I Thu-20:11:12 - #### Devices ####
I Thu-20:11:12 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/devices/area_detector.py
I Thu-20:11:12 - /home/mintadmin/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/devices/calculation_records.py
[TerminalIPythonApp] WARNING | Unknown error in handling startup files:
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/device.py in __get__(self, instance, owner)
    222         try:
--> 223             return instance._signals[self.attr]
    224         except KeyError:

KeyError: 'acquire'

During handling of the above exception, another exception occurred:

TimeoutError                              Traceback (most recent call last)
~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/IPython/core/shellapp.py in _exec_file(self, fname, shell_futures)
    376                     else:
    377                         # default to python, even without extension
--> 378                         self.shell.safe_execfile(full_filename,
    379                                                  self.shell.user_ns,
    380                                                  shell_futures=shell_futures,

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/IPython/core/interactiveshell.py in safe_execfile(self, fname, exit_ignore, raise_exceptions, shell_futures, *where)
   2747             try:
   2748                 glob, loc = (where + (None, ))[:2]
-> 2749                 py3compat.execfile(
   2750                     fname, glob, loc,
   2751                     self.compile if shell_futures else None)

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/IPython/utils/py3compat.py in execfile(fname, glob, loc, compiler)
    166     with open(fname, 'rb') as f:
    167         compiler = compiler or compile
--> 168         exec(compiler(f.read(), fname, 'exec'), glob, loc)
    169 
    170 # Refactor print statements in doctests.

~/.ipython-bluesky/profile_bluesky/startup/00_bluesky.py in <module>
     10     )
     11 )
---> 12 from instrument.collection import *

~/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/collection.py in <module>
     13 
     14 logger.info("#### Devices ####")
---> 15 from .devices import *
     16 
     17 logger.info("#### Callbacks ####")

~/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/devices/__init__.py in <module>
      5 # from .aps_source import *
      6 
----> 7 from .area_detector import *
      8 from .calculation_records import *
      9 from .ioc_stats import *

~/Documents/projects/BCDA-APS/bluesky_instrument_training/instrument/devices/area_detector.py in <module>
    110 
    111 
--> 112 adsimdet = MySimDetector("ad:", name="adsimdet", labels=("area_detector",))
    113 adsimdet.wait_for_connection(timeout=15)
    114 adsimdet.hdf1.create_directory.put(-5)

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/areadetector/trigger_mixins.py in __init__(self, image_name, *args, **kwargs)
    114 
    115     def __init__(self, *args, image_name=None, **kwargs):
--> 116         super().__init__(*args, **kwargs)
    117         if image_name is None:
    118             image_name = '_'.join([self.name, 'image'])

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/areadetector/trigger_mixins.py in __init__(self, *args, **kwargs)
     93                                     ('cam.image_mode', 1),  # 'Multiple' mode
     94                                     ])
---> 95             self._acquisition_signal = self.cam.acquire
     96 
     97         self._status = None

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/device.py in __get__(self, instance, owner)
    223             return instance._signals[self.attr]
    224         except KeyError:
--> 225             return instance._instantiate_component(self.attr)
    226 
    227     def __set__(self, instance, owner):

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/device.py in _instantiate_component(self, attr)
   1216 
   1217         try:
-> 1218             self._signals[attr] = cpt.create_component(self)
   1219             sig = self._signals[attr]
   1220             for event_type, functions in cpt._subscriptions.items():

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/device.py in create_component(self, instance)
    180         if self.lazy and hasattr(self.cls, 'wait_for_connection'):
    181             if getattr(instance, 'lazy_wait_for_connection', True):
--> 182                 cpt_inst.wait_for_connection()
    183 
    184         return cpt_inst

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/signal.py in wait_for_connection(self, timeout)
   1488     def wait_for_connection(self, timeout=1.0):
   1489         '''Wait for the underlying signals to initialize or connect'''
-> 1490         self._ensure_connected(self._read_pv, self._write_pv, timeout=timeout)
   1491 
   1492     @property

~/Apps/anaconda/envs/class_2021_03/lib/python3.8/site-packages/ophyd/signal.py in _ensure_connected(self, timeout, *pvs)
   1059         # @raise_if_disconnected can cause issues otherwise.
   1060         if not self._signal_is_ready.wait(timeout):
-> 1061             raise TimeoutError(f'Control layer {self.cl.name} failed to send connection and '
   1062                                f'access rights information within {float(timeout):.1f} sec')
   1063 

TimeoutError: Control layer pyepics failed to send connection and access rights information within 1.0 sec
```

</details>

prjemian, 2021-02-26:
How can `Component` get the intended timeout value to supply in the kwarg when calling `cpt_inst.wait_for_connection()`?

prjemian, 2021-02-26:
BTW, when I modify the local `ophyd/device.py` file, adding a `timeout=15` kwarg, the problem is fixed.

The only problem with fixing is this is to identify the right (connection) timeout value to use.

danielballan, 2021-02-26:
It looks like this was fixed on 12 January by @mrakitin in https://github.com/bluesky/ophyd/commit/7bcd868716319068e7e83afc2231c61e98049b92 but not yet released.The last ophyd release was 16 December. Someone from DAMA will release today, and we'll get back on our monthly cadence.

mrakitin, 2021-02-26:
Thanks @danielballan, yes, see https://github.com/bluesky/ophyd/pull/959 for the full story.

prjemian, 2021-02-26:
Great catch!  I'll close this now as a duplicate of #959.

closed 2021-02-26 by prjemian

prjemian, 2021-02-26:
BTW, the fix in #959 was to change the default timeout in `ophyd.signal.EpicsSignalBase.wait_for_connection()` from `1.0` to `DEFAULT_CONNECTION_TIMEOUT`.