bluesky/ophyd #981: Clearing pyepics callbacks for an ophyd object

bsobhani, 2021-03-19:
Back in December I remember looking for a way to delete the pyepics cache created by an ophyd object, once you are done with the ophyd object. It has been a while but I remember trying the destroy method, clear_callbacks, and release_pvs and none of them seemed to do what I wanted, which was clear the cache of a specific ophyd object while leaving the others in tact. So I wrote this function which seems to work:

https://github.com/bsobhani/bsstudio/blob/f05a34f583bf29488cbeb988ad7bdc5256216e96/bsstudio/functions.py#L232

Maybe the EpicsSignal class would benefit from having a method that clears the callbacks like this? Otherwise, you cannot create an ophyd object that is temporary, because this pyepics cache memory remains after the ophyd object goes out of scope, and I have seen it add up to large numbers over time.

jherkenhoff, 2024-12-14:
Same for me. For example, when deleting a device (`del myDevice`), the callbacks still get called

tacaswell, 2024-12-16:
@jherkenhoff What is your usecase for temporary ophyd objects?  In general the philosophy of the library is keep objects (and hence the circuits and channels they are sitting on top of) alive for the duration of the process.

jherkenhoff, 2024-12-28:
Actually no real usecase. I just stumbled upon it when debugging my custom callback logic and this behaviour felt like a serious "memory-leak" type behaviour (although not strictly a memory leak). This behaviour is not transparent to the user. When deleting an object, I do not suspect any code to ever be executed from this object again, but this is exactly what happens.