bluesky/ophyd #983: cannot find online documentation for PVPositionerPC

prjemian, 2021-04-30:
Expected to find this on [Positioners](https://blueskyproject.io/ophyd/positioners.html) page.

ambarb, 2023-09-26:
@prjemian @tacaswell is this where it would go?  

I am thinking i can copy paste and add some information about put_complete.  I forgot PC was referring to EPICS put_complete and we wasted a bunch of time troubleshooting an ophyd device because `caput` worked (where as `caput -c` would have shown something was wrong with the network traffic.

Thanks,
Andi

prjemian, 2023-09-26:
That's the class.  It does not show up in the online docs for probably some technical reason.

prjemian, 2023-09-26:
@ambarb : Any documentation contribution is welcome here.  I'm guessing the reason why the documentation in this module does not appear in the online docs for `PVPositionerPC()` is the lack of a docstring.