bluesky/ophyd #986: DOC publishing fails due to Sphinx error

prjemian, 2021-05-04:
The publishing build failed here:

```python
make: Entering directory '/home/runner/work/ophyd/ophyd/docs'
sphinx-build -b html -d build/doctrees  -W source build/html
Running Sphinx v1.6.7
making output directory...

Exception occurred:
  File "/opt/hostedtoolcache/Python/3.8.9/x64/lib/python3.8/site-packages/matplotlib/sphinxext/plot_directive.py", line 280, in setup
    app.add_css_file('plot_directive.css')
AttributeError: 'Sphinx' object has no attribute 'add_css_file'
The full traceback has been saved in /tmp/sphinx-err-hze40yz5.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
make: *** [Makefile:61: html] Error 1
make: Leaving directory '/home/runner/work/ophyd/ophyd/docs'
Error: Process completed with exit code 2.
```

The Sphinx version is exceptionally old (most recent release is 4.0.0b or some such).  The change this PR contributed cannot be responsible for that problem.  This is a new bug then.

_Originally posted by @prjemian in https://github.com/bluesky/ophyd/issues/984#issuecomment-831647029_

prjemian, 2021-05-04:
https://github.com/bluesky/ophyd/runs/2493817187?check_suite_focus=true

mrakitin, 2021-05-04:
That's what I found regarding the `AttributeError: 'Sphinx' object has no attribute 'add_css_file'` error: https://github.com/sphinx-doc/sphinx/issues/7747#issuecomment-636023612.

The sphinx package in ophyd's doc requirements has that hard pin:
https://github.com/bluesky/ophyd/blob/82053af306815a4980aa4689a31d717f3234bd43/requirements-docs.txt#L6

Matplotlib extension seems to require the latest API, so we should upgrade the sphinx version.

prjemian, 2021-05-18:
I'm using Sphinx 3.? or higher on my side. If the server builds the docs, I'm OK with closing this issue.

mrakitin, 2021-05-19:
Closed via #991.

closed 2021-05-19 by mrakitin