bluesky/ophyd #990: SignalRO not imported by ophyd/__init__.py

pshafer-als, 2021-05-15:
To use `SignalRO`, it must be imported from its submodule, rather than the main `ophyd` module.

```python
from ophyd.signal import SignalRO  # This works
from ophyd import SignalRO  # This does NOT work
```

Could `SignalRO` be imported in '[ophyd/__init__py](https://github.com/bluesky/ophyd/blob/master/ophyd/__init__.py#L78)' just like (`Signal`, `EpicsSignal`, `EpicsSignalRO`, `DerivedSignal`) and many other ophyd classes?  Or is `SignalRO` obscured, and intended only for edge cases?

'[ophyd/__init__py](https://github.com/bluesky/ophyd/blob/master/ophyd/__init__.py#L78)':
```python
[line 78] from .signal import (Signal, EpicsSignal, EpicsSignalRO, DerivedSignal)  # Replace this line...
[line 78] from .signal import (Signal, SignalRO, EpicsSignal, EpicsSignalRO, DerivedSignal)  # ...with this line
```

mrakitin, 2021-05-17:
@pshafer-als, thanks for the suggestion. I don't see any issue with it being available from the top-level import. A PR is always welcome.

pshafer-als, 2021-05-17:
Thanks for confirming.  I’ll submit a PR then.

-Padraic

commit 078fd5a1 (2021-05-19): Fix #990: Import 'SignalRO' with 'ophyd'

`SignalRO` is now imported by 'ophyd.__init__.py' 
so that it is available when importing `ophyd`

```python
from ophyd import SignalRO  # This now works
from ophyd import *  # ...so does this
```
commit 61625815 (2021-05-19): Merge pull request #991 from pshafer-als/issue-990/import-signalro

Fix #990: Import 'SignalRO' with 'ophyd'
closed 2021-05-19 by mrakitin