bluesky/ophyd #1002: Process `stage_sigs` correctly for `hdf5.warmup()`

mrakitin, 2021-08-02:
When we with @oksanagit tried to warm up the `hdf5` on an AreaDetector 3-10 with the "Internal" trigger mode, the process did not recognize that mode from the `.cam.trigger_mode.enum_strs`:
```py
In [4]: Diag1_CamV.cam.trigger_mode.enum_strs                                                                                  
Out[4]: 
('Free Run',
 'Sync In 1',
 'Sync In 2',
 'Sync In 3',
 'Sync In 4',
 'Fixed Rate',
 'Software')
```
The [`warmup` method](https://github.com/bluesky/ophyd/blob/705fc9bc0fee41aaca806bb9d66caba7a7cd99e5/ophyd/areadetector/plugins.py#L928) on the HDF5 plugin class has the `Internal` value hard-coded, so there is no way to update if there. And, as it wasn't in the list of the accepted strings in the `enum_strs` above, ophyd passed it as `b'Internal'` to the pyepics layer, and then it failed as it expected an integer rather than a binary string. Oksana may provide a more complete traceback with the details.

tacaswell, 2021-08-02:
I think the best solution here is to subclass the hdf5 plugin and override the warmup method.

We can not use `stage` in `warmup` because the issue we are trying to avoid is a segfault in the IOC in staging!   It is also not safe to dig out of stage_sigs to get the right value to set the mode to as the camera may be used with an external trigger that will never come while warming up!

----

An even better solution for this would be to add the functionality to AD for plugins to ask their source what size array will be coming without having to push an array through and infer it!

mrakitin, 2021-08-03:
@tacaswell, do you agree that the current behavior with passing non-validated string (that is not listed in enums) to the `pyepics` layer?

prjemian, 2021-08-03:
> @tacaswell, do you agree that the current behavior with passing non-validated string (that is not listed in enums) to the `pyepics` layer?

Yikes!  Does not sound as if this would be 100% successful.  When I encountered this situation (`warmup()` was using an enumerated string value that was not defined for the particular detector the particular detector support), the solution was to overlay a custom method instead.

prjemian, 2021-08-03:
As @tacaswell said:
> An even better solution for this would be to add the functionality to AD for plugins to ask their source what size array will be coming without having to push an array through and infer it!

[AD actually has such a feature](https://github.com/BCDA-APS/apstools/issues/419#issuecomment-693010898) ([`LazyOpen` in `Stream` mode](https://github.com/areaDetector/ADCore/blob/master/RELEASE.md#ndpluginfile-5)) that eliminates the need entirely for a `warmup()` method.  The `LazyOpen` feature can be used in many situations.  Unfortunately not in the tomography use case we encountered at APS 2-BM-B with the MONA project a few years ago.  Somewhat related to https://github.com/bluesky/ophyd/issues/713.