bluesky/ophyd #1021: Is there an ophyd.EpicsSignalWO class?

prjemian, 2021-11-05:
EPICS PV fields such as `.PROC` are signals for the EPICS server to run a function.  They never return the value that you put to them.  This breaks the current model for an ophyd EpicsSignal that requires the get to reach the put value.

tacaswell, 2021-11-05:
No, but there probably should be.

prjemian, 2021-11-05:
The `.PROC` field was a lousy example.  Better is the [`.SYNC`](https://github.com/epics-modules/motor/blob/da3bfab4c118226719f756779cc3e84a3a405ddf/motorApp/MotorSrc/motorRecord.dbd#L799-L802) field from EPICS motor.  That's the one we repeatedly run into the problem of `another call to set() in progress`.

prjemian, 2021-11-05:
To be clear (in the implied request):

> The proposed `ophyd.EpicsSignalWO` would be a write-only signal that does not check the EPICS PV for a final value.  It only does a `caput` and returns without checking for an updated value from EPICS.

tacaswell, 2021-11-05:
Can you still get put-completion callbacks on the .SYNC or .PROC fields?

prjemian, 2021-11-05:
Good question but I do not know.  But I know who to ask.

prjemian, 2021-11-05:
Might be easier just to test it though.

tacaswell, 2021-11-05:
I mean, at the protocol level, I can not image that they would reject a `WriteNotifyRequest`, the question is if the IOC does anything meaningful about it!

tangkong, 2022-05-12:
I believe this is closed by #1044 , but I don't have the authority to close the issue

mrakitin, 2022-05-12:
Closed via #1044.

closed 2022-05-12 by mrakitin