bluesky/ophyd #1026: this function is not aware of "tiff squashing".

MehmetTopsakal, 2022-01-11:
https://github.com/bluesky/ophyd/blob/df60483867e521dcda83648756cab8fc7b80dd17/ophyd/areadetector/detectors.py#L77

At XPD, we always use "tiff squashing" There is a key called "number_of_sets". 

I do this change manually for py file here :/xx//site-packages/ophyd/areadetector/detectors.py

        #shape = (self.cam.num_images.get(),
                 #self.cam.array_size.array_size_y.get(),
                 #self.cam.array_size.array_size_x.get())
        #if proc plugin is being used
        shape = (self.number_of_sets.get(),
                 self.cam.array_size.array_size_y.get(),
                 self.cam.array_size.array_size_x.get())

tacaswell, 2022-01-11:
The quick-fix is to over-ride this at XPD/PDF, the long-term fix is to push this method down into the plugins.  You could imagine there being two

MehmetTopsakal, 2022-07-18:
@tacaswell on a local conda environment, I was manually editing /xx//site-packages/ophyd/areadetector/detectors.py as shown above. But this is not ideal. Because I cannot edit files such as "/nsls2/conda/envs/2022-2.3-py39-tiled/lib/python3.9/site-packages/ophyd/__init__.py". Is there a way to apply this fix without editing source code? Thanks 

[image]

tacaswell, 2022-07-18:
via a local subclass.