bluesky/ophyd #1043: Should we disable https://nsls-ii.github.io/ophyd?

mrakitin, 2022-05-09:
Looks like we don't automatically push the documentation to https://nsls-ii.github.io/ophyd, that may confuse users. Probably we should remove the projects that are published to blueskyproject.io from nsls-ii.github.io.

**Note:** the nsls-ii.github.io is used for other active NSLS-II projects.

danielballan, 2022-05-12:
Possibility solutions, in order of increasing effort:

1. Set up a single redirect from the old index page to the new index page, and let all other pages be 404.
2. Replace every old .html file with a page that gives a link to its the new index page. Nothing will 404, but you'll have to go and find the page you want.
3. Try to map the old page to the correct new page, if it still exists. Seamless, but sounds hard.

tacaswell, 2022-05-12:
https://github.com/matplotlib/matplotlib.github.com/blob/main/cleanout.py and https://github.com/matplotlib/matplotlib.github.com/blob/main/_websiteutils/make_redirects_links.py might go useful starting places for 2 and 3 respectively.