bluesky/ophyd #1056: blocking_callbacks in area detector setup

prjemian, 2022-07-27:
@MarkRivers noticed in [apstools' example for ADPilatus](https://bcda-aps.github.io/apstools/examples/_ad__pilatus.html):

```py
OrderedDict([('enable', 1),
            ('blocking_callbacks', 'Yes'),
```

> It is really not good practice to be setting blocking_callbacks to Yes.  That slows down the detector.  The preferred practice is to set blocking_callbacks=No in the plugin and set wait_for_plugins=Yes in the driver.  That way all plugins can run in their own threads without slowing down the driver, and the detector will still wait for them all to complete before it reports that acquisition is done.

The same observation [applies to ophyd](https://github.com/bluesky/ophyd/search?q=blocking_callbacks):

https://github.com/bluesky/ophyd/blob/aa17c1e896aa3874ff722d72c8f6d9aa59c3ebaa/ophyd/areadetector/plugins.py#L172

and [`area-detector.rst`](https://github.com/bluesky/ophyd/blob/0895f9f00bdf7454712aa954ea7c7f3f1776fcb9/docs/source/explanations/area-detector.rst#L54)

Thanks!  I'll label this as a _bug_ here.

prjemian, 2022-07-27:
I do not see `WaitForPlugins` in a code search: https://github.com/bluesky/ophyd/search?q=WaitForPlugins, just issue content.

No mention in apstools: https://github.com/BCDA-APS/apstools/search?q=WaitForPlugins

BTW, APS use will derive from the ophyd repo, so that is where the feature would be provided.  The documentation in apstools will be revised.

MarkRivers, 2022-07-27:
The documentation here ([area-detector.rst](https://github.com/bluesky/ophyd/blob/0895f9f00bdf7454712aa954ea7c7f3f1776fcb9/docs/source/explanations/area-detector.rst)) explains that the new scheme using WaitForPlugins is supported by doing the following:

*****************************
To convert an existing area detector sub-class to support the new scheme you must:

- Change the type of the came to sub-class [:obj:`nslsii.ad33.CamV33Mixin`](https://github.com/bluesky/ophyd/blob/0895f9f00bdf7454712aa954ea7c7f3f1776fcb9/docs/source/explanations/area-detector.rst#id4)
- Change the trigger mixin to be [:obj:`nslsii.ad33.SingleTriggerV33`](https://github.com/bluesky/ophyd/blob/0895f9f00bdf7454712aa954ea7c7f3f1776fcb9/docs/source/explanations/area-detector.rst#id6)
- Arrange for det.cam.ensure_nonblocking to be called after initializing the ophyd object.

tacaswell, 2022-07-27:
@MarkRivers 's analysis is correct (https://blueskyproject.io/ophyd/explanations/area-detector.html#callbacks is the rendered version of those docs).

We took a pass at trying to make the AD classes smart enough to detect the AD version and use the correct ohpyd classes (and @klauer put in the work to label the plugin classes with the supported versions), however that work stalled.

It has been long enough we probably should assume that we always have access to AD3-3 + and make the default classes reflect that, but it will require a champion to lead that.