bluesky/ophyd #1059: Update build system

coretl, 2022-08-02:
To support the development of Ophyd.v2, some changes will be made. This issue outlines a series of steps that will make these changes. Each change will be a separate PR. Please comment on this ticket if you would like to discuss any of the changes. I will tag members of the Technical Steering Committee in each PR so they have a chance to review before merge.

The steps:

1. Adopt black for code formatting, and use pre-commit to run black and flake8 on checkin and in CI. Blacken the whole codebase in a single commit. Put that commit in `.git-blame-ignore-revs` so it can be [ignored](https://docs.github.com/en/repositories/working-with-files/using-files/viewing-a-file#ignore-commits-in-the-blame-view)
2. Adopt isort for import ordering, run as a plugin to flake8 using `flake8-isort`. Adopt in a single commit as per black.
3. Move config from `.flake8`, `pytest.ini`, `setup.py`, `coveragerc`, and `requirements*.txt` to `setup.cfg` and adopt setuptools-scm instead of versioneer (which will also allow us to delete `MANIFEST.in`)
4. Use `pytest-cov` and `codecov` instead of `coverage` commands
5. Add `.vscode` settings to source control
6. Move the codebase from `ophyd` to `ophyd/v1` and add an `ophyd/__init__.py` that import the v1 namespace. New code will be added under the `ophyd/_v2` namespace until is stabilizes
7. Start creating v2 docs, but not decided where to put them in the docs structure yet

@danielballan @tacaswell @mrakitin @klauer @ZLLentz @prjemian @dylanmcreynolds @whs92 @coretl @callumforrester

dylanmcreynolds, 2022-08-02:
I am a vscode user and  I have been a part of projects that add .vscode to the project and those that add it to .gitignore. I prefer the .gitignore method, myself, as I use strange unknown extensions that most people wouldn't care about. I don't feel too strongly here, but is this really needed? Or is there a way to still have personal vscode settings while drawing some settings from the repo?

prjemian, 2022-08-02:
Editor settings are not part of the code base.  In a repository, they become cruft as editor choice evolves.

coretl, 2022-08-03:
That sounds like consensus to me, I won't add editor settings in. Any comments on the rest of the steps?

coretl, 2022-08-03:
Can you think of an example where module level properties have been done in another project? Save me reinventing the wheel...

tacaswell, 2022-08-03:
https://github.com/bluesky/databroker/blob/f699d576c6547df1b8ef822dc51d86b61e1b03cf/databroker/__init__.py#L48-L79 is in databroker

A helper class we might want to vendor from Matplotlib https://github.com/matplotlib/matplotlib/blob/ffc55ae5c3b62b36fd994800404fbad98cb9e853/lib/matplotlib/_api/__init__.py#L195-L227

https://scientific-python.org/specs/spec-0001/ <- general discussion of lazy loading

https://pypi.org/project/lazy_loader/ <- a whole package to make this easier

https://discuss.python.org/t/pep-690-lazy-imports/15474 <- pep about making lazy loading part of the language to make all of this un-needed

prjemian, 2022-10-11:
I'm ok with the other points. Thanks for describing.