bluesky/ophyd #1063: Add contributing guide

coretl, 2022-08-05:
As I'm going to be making a lot of contributions to Ophyd, it would be good to nail down how contributors should work with the repository. I'm thinking something with the same type of content as [this](https://dls-controls.github.io/scanspec/master/reference/contributing.html), except I think it is probably a how-to guide titled "How to contribute to the project".

Does something similar already exist? Or are there notes anywhere on how to run the tests with dependencies? Otherwise I will take the CI job as the way people run things locally and adapt the instructions to suit

danielballan, 2022-09-26:
I have a memory of documentation about starting up Docker and running the tests, but I cannot find it. In any case it would short and seems like to become outdated in the immediate future so I think we can carry on with a fresh start.

coretl, 2022-09-26:
It's still there:
https://blueskyproject.io/ophyd/how-to/docker.html

I'll work that into the contributing guide when I get to it