bluesky/ophyd #1072: Need help in an error: ophyd.utils.errors.DisconnectedError

magiconline, 2022-09-13:
I had this error when I want to write some demo.

m2 and current are physical devices and can be access through terminal and pyepics. If I run this code in a .py file such as `/home/cz/anaconda3/envs/AIECS/bin/python /home/cz/AIECS/src/bluesky_demo.py`, sometimes there will be an error  `ophyd.utils.errors.DisconnectedError`.  And it always happens after a successful run.

But if I run the same code in jupyter notebook, it always works fine. How can I fix this?

Code:
```
from bluesky import RunEngine
from bluesky.callbacks.best_effort import BestEffortCallback
from bluesky.plans import scan, rel_scan, count
from ophyd import EpicsMotor
from ophyd.signal import EpicsSignalRO

m2 = EpicsMotor('X08U1:OP:M2:TRX', name='m2')
current = EpicsSignalRO('X08U1A:EX:Aperture:PA:Current', name='current')

RE = RunEngine({})
bec = BestEffortCallback()
RE.subscribe(bec)

RE(count([m2, current], 10))
```

Output:
```
Transient Scan ID: 1     Time: 2022-09-13 19:29:32
Persistent Unique Scan ID: 'c6a571ed-ea4b-4679-a3a6-e2ba681c2ffe'

Run aborted
Traceback (most recent call last):
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 1365, in _run
    msg = self._plan_stack[-1].send(resp)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plans.py", line 77, in count
    return (yield from inner_count())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/utils.py", line 1127, in dec_inner
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 952, in stage_wrapper
    return (yield from finalize_wrapper(inner(), unstage_devices()))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 509, in finalize_wrapper
    ret = yield from plan
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 950, in inner
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/utils.py", line 1127, in dec_inner
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 325, in run_wrapper
    yield from contingency_wrapper(plan,
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 575, in contingency_wrapper
    ret = yield from plan
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plans.py", line 74, in inner_count
    return (yield from bps.repeat(partial(per_shot, detectors),
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 1183, in repeat
    return (yield from repeated_plan())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 1166, in repeated_plan
    yield from ensure_generator(plan())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 1026, in one_shot
    yield from take_reading(list(detectors))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 911, in trigger_and_read
    return (yield from rewindable_wrapper(inner_trigger_and_read(),
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 693, in rewindable_wrapper
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 905, in inner_trigger_and_read
    reading = (yield from read(obj))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 95, in read
    return (yield Msg('read', obj))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 1425, in _run
    new_response = await coro(msg)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 1670, in _read
    ret = obj.read(*msg.args, **msg.kwargs)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/ophyd/device.py", line 1237, in read
    res.update(component.read())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/ophyd/utils/epics_pvs.py", line 200, in wrapper
    raise DisconnectedError('{} is not connected'.format(self.name))
ophyd.utils.errors.DisconnectedError: m2 is not connected

Traceback (most recent call last):
  File "/home/cz/AIECS/src/bluesky_demo.py", line 18, in <module>
    RE(count([m2, current], 10))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 807, in __call__
    self._resume_task(init_func=_build_task)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 931, in _resume_task
    raise exc
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 1500, in _run
    raise err
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 1365, in _run
    msg = self._plan_stack[-1].send(resp)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plans.py", line 77, in count
    return (yield from inner_count())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/utils.py", line 1127, in dec_inner
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 952, in stage_wrapper
    return (yield from finalize_wrapper(inner(), unstage_devices()))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 509, in finalize_wrapper
    ret = yield from plan
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 950, in inner
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/utils.py", line 1127, in dec_inner
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 325, in run_wrapper
    yield from contingency_wrapper(plan,
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 575, in contingency_wrapper
    ret = yield from plan
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plans.py", line 74, in inner_count
    return (yield from bps.repeat(partial(per_shot, detectors),
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 1183, in repeat
    return (yield from repeated_plan())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 1166, in repeated_plan
    yield from ensure_generator(plan())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 1026, in one_shot
    yield from take_reading(list(detectors))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 911, in trigger_and_read
    return (yield from rewindable_wrapper(inner_trigger_and_read(),
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/preprocessors.py", line 693, in rewindable_wrapper
    return (yield from plan)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 905, in inner_trigger_and_read
    reading = (yield from read(obj))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/plan_stubs.py", line 95, in read
    return (yield Msg('read', obj))
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 1425, in _run
    new_response = await coro(msg)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/bluesky/run_engine.py", line 1670, in _read
    ret = obj.read(*msg.args, **msg.kwargs)
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/ophyd/device.py", line 1237, in read
    res.update(component.read())
  File "/home/cz/anaconda3/envs/AIECS/lib/python3.8/site-packages/ophyd/utils/epics_pvs.py", line 200, in wrapper
    raise DisconnectedError('{} is not connected'.format(self.name))
ophyd.utils.errors.DisconnectedError: m2 is not connected
```

tacaswell, 2022-09-13:
I would try adding 

```python
m2.wait_for_connection()
```

before starting the plan.  We let the epics PVs connect in the background (so that big profiles start fast(ish)).  I think because you have this in a script you are sometimes losing the race between starting the scan and the motor finishing its connection.

The reason it works fine in the notebook is I suspect you have these things in different cells and the time it takes you to run the cells (and the time it takes the jupyter kernel to negotiate with the front end etc) is enough that you always win the race.  Thus by the time you try to use it the motor is actually connected.

magiconline, 2022-09-14:
It works. Thanks.

closed 2022-09-14 by magiconline