bluesky/ophyd #1075: abstracted interface for area detectors

xiaoqiangwang, 2022-09-13:
As I understand the current area detectors support (*ADBase*, *CamBase*) are modelled after epics [areaDetector](https://github.com/areaDetector/areaDetector) module. This certainly makes sense because areaDetector module is de facto standard in integrating area detectors.

If 
Looking at the *PositionerBase* interface to motors, an abstracted interface, a minimal set of the operations for detectors, like the motor vs .

danielballan, 2022-09-16:
Thanks @xiaoqiangwang. This is a very sensible suggestion.

danielballan, 2022-09-21:
With a little more time to think, let me respond more thoroughly.

Acquire start/stop is already handled by the [Triggerable protocol](https://blueskyproject.io/bluesky/hardware.html#bluesky.protocols.Triggerable), which initiates "start" and returns [status object](https://blueskyproject.io/ophyd/explanations/status.html) that provides a way to tell when it has stopped.

Stnadardizing exposure time time is harder. In the EPICS areaDetector module there are several relevant parameters ([ADAcquireTime, ADAcquirePeriod, ADNumImages, ADNumExposures](https://cars9.uchicago.edu/software/epics/areaDetectorDoc.html)). This may not generalize beyond areaDetector, as different detectors may parameterize these choices differently.

xiaoqiangwang, 2022-09-22:
Thanks for being positive. Agreed that it is a difficult design decision.

It does not have to be abstracted _beyond_ areaDetector. After all, areaDetector is an effort from Mark Rivers to abstract area detector operations. The operations under "Collect" and "Readout" tabs could be in the abstracted interface. If the actual detector does not support, say ADNumExposures, it can either simulate the accumulation or always set it to 1.
[image]

[image]

Actually the current CamBase is such an interface, but EPICS centric. I imagine the following abstraction diagram
```
                                    Device
                                       |
                         AbstractNDDetector (mostly abstract properties )
                                      |
              |----------------------------------|
  CamBase (areaDetector)                     Other camera not in areaDetector
             |
|------------|------------------|--
ADAravis    ADSimDetector      
```