bluesky/ophyd #1108: Switch EPICS port at runtime

DominicOram, 2023-04-17:
At DLS we run a simulated beamline on different EPICS ports to the real beamline. We would like a smoke test where we run up our ophyd devices and confirm they connect to the real beamline and the fake beamline. I ran across problems with this as I could find no easy way to switch the ports ophyd was using at runtime. The quick solution I have is just to run the tests one after another in different processes, which is fine. But is there a way of doing this more cleanly in `ophyd` or potential to add it in to `ophyd v2`?

coretl, 2023-04-17:
Both v1 and v2 use `libca` under the hood, where the only way to change is via environments variables: https://epics.anl.gov/base/R7-0/7-docs/CAref.html#EPICS

However, I'm not sure if disconnecting the channels, setting the environment variable, then reconnecting the channels might use a new environment variable? If so, then we could find a way to expose that functionality in ophyd v2

prjemian, 2023-04-17:
> Is there an easy way to disconnect all channels? I assume there's some
> cache that you have to clear? As I said, this isn't that big a deal so if
> it's not easily possible it's fine

Post this question to EPICS ***@***.***

prjemian, 2023-04-17:
In your message, day what you are really doing (that you are testing an
EPICS client first on one port, then on a non standard port). I'm thinking
you might need to run two different test threads to do this.

tacaswell, 2023-04-17:
The cache is shared between https://github.com/bluesky/ophyd/blob/master/ophyd/_pyepics_shim.py and pyepics (if you are using the pyepcis version not the caproto version).

I would not trust doing simulated and real PVs in the same test process.  Even if you could manage to kill them all reliably _now_, details like exactly how caches work is not something I would reasonably think that ophyd, pyepics, or caproto would guarantee to be stable (or even be part of the public API) so would have no faith that that method would work in the future.  I have grave concerns that eventually you are going to end up running your test suite against real hardware while a user is running.....

DominicOram, 2023-04-18:
We're running through a gateway and so only have read access to the real PVs but I agree that it's some risk. It sounds like the answer may be that changing ports at runtime is outside of the scope of what ophyd can and will ever do. Which is fine and I am happy to close this off with that conclusion.

tacaswell, 2023-04-18:
I'm going to close this as deferred until libca supports it.

closed 2023-04-18 by tacaswell