bluesky/ophyd #1121: Helper function to chain functions that return statuses

DominicOram, 2023-05-30:
On MX at DLS we have a usecase where we want to kickoff a series of tasks on the device that must be done in series e.g.
```python
class MyDevice(Device):
   def do_1(self) -> Status:
      ...
   def do_2(self) -> Status:
      ...
   # Some more do_x
    
   def do_all(self) -> Status:
      status_1 = self.do_1()
      status_1.add_callback(self.do_2)
      ....
      return status_x
```
@olliesilvester has written some code for this but we think it may be generically useful to be in ophyd so they will add a PR for it here.

tacaswell, 2023-06-06:
Could this sequencing be lifted up to be a plan that the RE manages?

coretl, 2023-06-07:
> ```python
> def my_async_plan():
>     yield from bps.set(.., group="A")
>     yield from bps.wait("A") # Note this shouldn't block main thread just this plan
>     ...
>     
> def my_actual_plan():
>     yield from bps.run_async_plan(my_async_plan, group="async_plan") # New message that's basically create a thread
>     yield from bps.set(.., group="B")
>     yield from bps.wait("B") # Should not block async thread
>     ...
>     yield from bps.wait("async_plan") # Effectively a join here
> ```

@warejosephb has written a task graph implementation that solves this for sub plans that end by returning a status object. I might have got this wrong, but I think you can construct a task graph that says "do A,B,C in parallel, then do D", and if A, B and C all do some setup and return a Status object, then it will do them all in quick succession then wait for the 3 Status objects, then do D.

> On MX at DLS we have a usecase where we want to kickoff a series of tasks on the device that must be done in series

For things that are "prepare for and do a scan" I think we are covered with `kickoff` and `complete` doing the sequencing, and multiple flyers for the distinct Devices taking part in the scan. However for arbitrary sequencing of unrelated events it is not so clear. If we wanted to do a background robot load while simultaneously doing an alignment scan then it would sound like the tasks should actually be independent.

What's your use case?

tacaswell, 2024-01-03:
I meant to link to https://github.com/bluesky/bluesky/pull/541 above