bluesky/ophyd #1122: Improve how EpicsMotor deals with `.velocity` and `.acceleration`

tacaswell, 2023-06-06:
Lifting a comment from slack from @dperl-dls .

If you try to set the velocity of an EpicsMotor out of bounds via `yield from bps.mv(mtr.velocity, target)` it will hang (I assume that the IOC clips the velocity to its valid range).

2. set "sensible" limit on the velocity and acceleration components of EpicsMotor
3. make at least the read and write timeouts runtime mutable
4. write a EpicsMotorVelocity(EpicsSignal) class that knows how to dig out VMAX (and add VMAX to EpicsMotor)

coretl, 2023-08-02:
I believe if you do a caput callback on the velocity it will return quickly, then you can caget the value to see whether it got clipped...

tacaswell, 2023-08-02:
Moving to using put-complete + checking is a better option than 1 and 2 and complimentary with 3.