bluesky/ophyd #1132: V2 'Simple Device' Tutorial Issue

oscarbranson, 2023-07-06:
New to ophyd, and immediatlely hit a problem... The tutorial to [Make a Simple Device](https://blueskyproject.io/ophyd/user_v2/how-to/make-a-simple-device.html) appears to be broken.

Minimal example taken from the source:

```python
"""Demo EPICS Devices for the tutorial"""

from enum import Enum

from ophyd.v2.core import StandardReadable
from ophyd.v2.epics import epics_signal_r, epics_signal_rw

class EnergyMode(Enum):
    """Energy mode for `Sensor`"""

    #: Low energy mode
    low = "Low Energy"
    #: High energy mode
    high = "High Energy"

class Sensor(StandardReadable):
    """A demo sensor that produces a scalar value based on X and Y Movers"""

    def __init__(self, prefix: str, name="") -> None:
        # Define some signals
        self.value = epics_signal_r(float, prefix + "Value")
        self.mode = epics_signal_rw(EnergyMode, prefix + "Mode")
        # Set name and signals for read() and read_configuration()
        self.set_readable_signals(
            read=[self.value],
            config=[self.mode],
        )
        super().__init__(name=name)
        
s = Sensor('test')
```

yields:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[2], line 31
     25         self.set_readable_signals(
     26             read=[self.value],
     27             config=[self.mode],
     28         )
     29         super().__init__(name=name)
---> 31 s = Sensor('test')

Cell In[2], line 22, in Sensor.__init__(self, prefix, name)
     20 def __init__(self, prefix: str, name="") -> None:
     21     # Define some signals
---> 22     self.value = epics_signal_r(float, prefix + "Value")
     23     self.mode = epics_signal_rw(EnergyMode, prefix + "Mode")
     24     # Set name and signals for read() and read_configuration()

File [~/.python/py3/lib/python3.11/site-packages/ophyd/v2/epics.py:87](https://file+.vscode-resource.vscode-cdn.net/home/oscar/CamDrive/bluesky/~/.python/py3/lib/python3.11/site-packages/ophyd/v2/epics.py:87), in epics_signal_r(datatype, read_pv)
     77 def epics_signal_r(datatype: Type[T], read_pv: str) -> SignalR[T]:
     78     """Create a `SignalR` backed by 1 EPICS PV
     79 
     80     Parameters
   (...)
     85         The PV to read and monitor
     86     """
---> 87     backend = _make_backend(datatype, read_pv, read_pv)
     88     return SignalR(backend)

File [~/.python/py3/lib/python3.11/site-packages/ophyd/v2/epics.py:56](https://file+.vscode-resource.vscode-cdn.net/home/oscar/CamDrive/bluesky/~/.python/py3/lib/python3.11/site-packages/ophyd/v2/epics.py:56), in _make_backend(datatype, read_pv, write_pv)
     54 w_transport, w_pv = _transport_pv(write_pv)
     55 transport = get_unique({read_pv: r_transport, write_pv: w_transport}, "transports")
---> 56 return transport.value(datatype, r_pv, w_pv)

AttributeError: 'function' object has no attribute 'value'
```

Running ophyd '1.8.0'.

coretl, 2023-07-07:
The problem is that ophyd v2 supports both CA and PVA, so the base install doesn't install either. You need to `pip install ophyd[ca]` to get the CA libraries. Unfortunately this isn't added in the tutorials, I have done so in https://github.com/bluesky/ophyd/pull/1133. The error message is bad because of https://github.com/bluesky/ophyd/issues/1113 which was fixed a few days ago but hasn't been released yet. To test this theory, please can you do the following:

- Post the output of `pip freeze`
- Do a `pip install git+https://github.com/bluesky/ophyd.git` and run the tutorial code again, and check you get a better error message saying something like "CA support not available"
- Do a `pip install ophyd[ca]` and the tutorial should work

oscarbranson, 2023-07-31:
Thanks! It would be useful to include some mention of this in the bluesky installation docs.

closed 2023-07-31 by oscarbranson