bluesky/ophyd #1143: Allow setting FileStoreBase formatter, so that write_path_template, read_path_template can use non-strftime formats

DiamondJoseph, 2023-08-04:
At Diamond, we want our raw files written into directories for the current visit, and would ideally represent the exactly location for the file to land in our ResourceDocuments.

This will probably end up being the in the root of the document rather than the resourcePath, such that the resourcePath is the path relative to the visit directory, but being able to form different paths inside the visit may be useful.

prjemian, 2023-08-04:
Help me understand this more easily.

Because we had (a couple years ago), a set of NeXus/HDF5 master:external
files from the DLS small-angle beam line where the external file link was
expressed in absolute directory form. Could not open the data since we did
not have the DLS file system was described in the file. (We talked with the
beamline about this and they responded.)

This is the type of problem you are trying to avoid. Please describe your
implementation again.

DiamondJoseph, 2023-08-04:
This ticket is mostly not connected to root/resourcePath handling, which I'm actively trying to avoid falling into your mentioned issue with.

root will be <some/diamond/specific/filesystem/path>/visit
resourcePath will be .</some/structure/within/a/visit>
.nxs files will only have ./resourcePath links written into them.

But our resourcePath/structure within a visit probably wants to be something like ./runN/, rather than ./2023 or ./20230804
Our root can ~easily be set to a new visit directory on stage as appropriate for the run we are staging for, but our resourcePath seems limited to just strftime