bluesky/ophyd #1149: Codecoverage tool fails easily; some lines accidentally covered

RAYemelyanova, 2023-08-21:
In a recent PR https://github.com/bluesky/ophyd/pull/1148 The codecoverage patch job was failing, although all the lines added in the PR had been tested.

The coverage on master for `ophyd/ophyd/areadetector/paths.py` is currently 91.97%. However, on the PR this same file has a lower coverage, although the PR did not change any of the lines pertaining to these tests or the source code. In fact, the PR concerns itself with ophyd v2 ,and the test coverage drop is for ophyd v1 functionality.

On further inspection, the last latest ophyd release had a codecoverage report with the same code coverage of this file as the PR's code (seen here: https://app.codecov.io/gh/bluesky/ophyd/commit/51679551aab7798424124509d48f7a006e73ca15/blob/ophyd/areadetector/paths.py). It seems like whether or not lines 73-80 get covered in the codecov report is entirely random, and looking at the code it makes sense; this bit of code will only get called if a signal.set call doesn't immediately line up with the value one is trying to set it with. i.e. if for some reason setting the signal takes a bit of time, this block of code will get called.

So, this issue is to propose two fixes:

1. Add an actual test for these lines of code, so that the patch change for the above PR passes,
2. Introduce tolerances for code coverage (+/- 0.1%) within which the jobs will still pass. Considering the sheer volume of tests involved, this is not the first instance where someone changing bits of ophyd code has codecov failing even though they've covered all their new lines of codes in tests.

callumforrester, 2023-08-21:
Agree with the fixes, but do we know why this is happening? I don't think I've seen it on other repositories.

RAYemelyanova, 2023-08-21:
The percentage decrease in patch is, in my opinion, totally random. I'm sure if I re-ran the CI on master a few times I'd get one CI job where (if coverage were also rerun, and the PR that this issue arose from) the patch change would be 0%.

The offending lines are these:

```python
    signal.put(val)
    deadline = ttime.time() + timeout if timeout is not None else None
    current_value = signal.get()

    while not path_compare(current_value, val, semantics=path_semantics):
        logger.debug(
            "Waiting for %s to be set from %r to %r...", signal.name, current_value, val
        )
        ttime.sleep(poll_time)
        if poll_time < 0.1:
            poll_time *= 2  # logarithmic back-off
        current_value = signal.get()
```

this is inside ophyd/areadetector/paths.py.

So basically the while loop is never entered if everything works smoothly. If things are a bit slow, however (e.g. runners are a bit slower than usual... takes a while for caput to actually complete) then the while loop might well be entered.

RAYemelyanova, 2023-08-21:
Just to confirm this, I'm going to rerun the CI on master a few times... hold on...

closed 2023-08-23 by RAYemelyanova