bluesky/ophyd #1180: Why is Cpt(EpicsMotor) and EpicMotor creating different configuration_attrs?

ambarb, 2024-03-12:
## Expected Behavior
[This simple EpicsMotor](https://github.com/NSLS-II-CSX/profile_collection/blob/f181f20cea984d07f9dbb471c0260faa90328f6a/startup/csx1/startup/endstation.py#L56) 
has the following descriptors:
```python

In [61]: list(db[-1].descriptors[0]['configuration']['es_diag1_y']['data'] )
Out[61]: 
['es_diag1_y_user_offset',
 'es_diag1_y_user_offset_dir',
 'es_diag1_y_velocity',
 'es_diag1_y_acceleration',
 'es_diag1_y_motor_egu']

```
This makes 100% sense to me.  I would not change this.

## Current Behavior
[But when EpicsMotor is used as a Cpt](https://github.com/NSLS-II-CSX/profile_collection/blob/f181f20cea984d07f9dbb471c0260faa90328f6a/startup/csx1/startup/tardis.py#L31), the default descriptors (configuration data)
is incomplete (in this case, pay attention to **tardis_theta**, **tardis_delta**, and **tardis_gamma** hint: the last 3 items in the list)
```python
In [3]: list(db[-1].descriptors[0]['configuration']['tardis']['data'] )
Out[3]: 
['tardis_energy',
 'tardis_energy_units',
 'tardis_energy_offset',
 'tardis_geometry_name',
 'tardis_class_name',
 'tardis_sample_name',
 'tardis_lattice',
 'tardis_lattice_reciprocal',
 'tardis_U',
 'tardis_UB',
 'tardis_reflections_details',
 'tardis_ux',
 'tardis_uy',
 'tardis_uz',
 'tardis_diffractometer_name',
 'tardis__hklpy_version',
 'tardis__pseudos',
 'tardis__reals',
 'tardis__constraints',
 'tardis__mode',
 'tardis_orientation_attrs',
 'tardis_theta_acceleration',
 'tardis_delta_user_offset',
 'tardis_gamma_velocity']

```
## Possible Solution
I can make it match as expected by iterating through a list and setting `.kind` to `"config"` but I wondering if there is a more sustainable way to do this without having to do anything inside the custom class

@prjemian maybe you are interested.  Did  you do something simliar as above?

## Steps to Reproduce (for bugs)
I think I provided enought links to a profile to get started

## Context
I noticed this because I want to correlate some decision making in which the "user" value is reset by changed the "user_offset" PV.  Others might want it to calculated scan overhead and see if a problem is develping over time, or it could be useful for flyscanning to know the velocity --- plus, this is a good for troubleshooting, expecially if these PV are easily changed in a GUI but there is no archiving of the values 

## Your Environment
`2022-2.1-py39-tiled`

tacaswell, 2024-03-12:
I moved it to ophyd.

My guess is that something either in the pseudo positioner code (I looked quickly but did not see anything), your profile, or in hkl is changing the kind of the missing (sub) components.

prjemian, 2024-03-16:
Seems more an issue for hklpy.  @ambarb Can you post a copy of the class where the Tardis and its real-space motors are defined?  Something that starts like this:

```py
class Tardis(E6C):
    theta = Component(EpicsMotor, ...
```

prjemian, 2024-03-16:
Simulating locally,

```py
class SixCircle(hkl.SimMixin, hkl.E6C):
    """
    Our 6-circle.  Eulerian.

    Energy obtained (RO) from monochromator.
    """

    # the reciprocal axes are defined by SimMixin

    mu = Component(EpicsMotor, f"{M_MU}", kind="hinted", labels=["motor"])
    omega = Component(EpicsMotor, f"{M_OMEGA}", kind="hinted", labels=["motor"])
    chi = Component(EpicsMotor, f"{M_CHI}", kind="hinted", labels=["motor"])
    phi = Component(EpicsMotor, f"{M_PHI}", kind="hinted", labels=["motor"])
    gamma = Component(EpicsMotor, f"{M_GAMMA}", kind="hinted", labels=["motor"])
    delta = Component(EpicsMotor, f"{M_TTH}", kind="hinted", labels=["motor"])

    energy = Component(EpicsSignalRO, "BraggERdbkAO", kind="hinted", labels=["energy"])
    energy_units = Component(EpicsSignalRO, "BraggERdbkAO.EGU", kind="config")

sixc = SixCircle(IOC, name="sixc")
sixc.wait_for_connection()
sixc._update_calc_energy()
```

The `delta` motor is PV `gp:m23`:

```py
delta = EpicsMotor("gp:m23", name="delta")
```

Compare them:

```py
In [5]: sixc.delta.read_configuration()
Out[5]: 
OrderedDict([('sixc_delta_user_offset',
              {'value': 0.0, 'timestamp': 1710425767.554572}),
             ('sixc_delta_acceleration',
              {'value': 0.2, 'timestamp': 1710425767.554572})])

In [9]: delta.read_configuration()
Out[9]: 
OrderedDict([('delta_user_offset',
              {'value': 0.0, 'timestamp': 1710425767.554572}),
             ('delta_user_offset_dir',
              {'value': 0, 'timestamp': 1710425767.554572}),
             ('delta_velocity',
              {'value': 1.0, 'timestamp': 1710425767.554572}),
             ('delta_acceleration',
              {'value': 0.2, 'timestamp': 1710425767.554572}),
             ('delta_motor_egu',
              {'value': 'degrees', 'timestamp': 1710425767.554572})])
```

prjemian, 2024-03-16:
Let's test what this looks like in a simpler `Device` class:

```py
In [12]: class Demo(Device):
    ...:     delta = Component(EpicsMotor, "gp:m23")
    ...: 

In [13]: demo = Demo("", name="demo")

In [14]: demo.delta.read_configuration()
Out[14]: 
OrderedDict([('demo_delta_user_offset',
              {'value': 0.0, 'timestamp': 1710425767.554572}),
             ('demo_delta_user_offset_dir',
              {'value': 0, 'timestamp': 1710425767.554572}),
             ('demo_delta_velocity',
              {'value': 1.0, 'timestamp': 1710425767.554572}),
             ('demo_delta_acceleration',
              {'value': 0.2, 'timestamp': 1710425767.554572}),
             ('demo_delta_motor_egu',
              {'value': 'degrees', 'timestamp': 1710425767.554572})])
```

prjemian, 2024-03-16:
Looks like some of the configuration is attenuated in the `Diffractometer` class.  Moving this to hklpy repo...