bluesky/ophyd #1196: Is the BadPVName exception enforcing an EPICS naming rule or just a convention?

gazzar, 2024-06-19:
I'm getting the BadPVName exception
`BadPVName: SR05ID01MIR01:t.Y.RBV`
raised when attempting a get() on an EpicsMotor.

Although most comments from colleagues suggest that this seems to be a poorly named PV, is it actually illegal in some way?
The original implementation comment suggests it's more of a guideline than a rule https://github.com/bluesky/ophyd/commit/1e973e162e926d67fbf73b1e03113888258da8e6
Our legacy code based on caget, which we still need to use, is quite happy to read it and I can't find any rule in the EPICS docs about dots in PV names being disallowed.

I think the current behaviour should be able to be bypassed if necessary if it is enforcing a convention rather than a rule.

We will try to create an alias of the PV for this case.

prjemian, 2024-06-19:
This code is improperly informed: https://github.com/bluesky/ophyd/blob/030c668da64443c54e17679f3342df371c52a620/ophyd/utils/epics_pvs.py#L74-L75

While very unusual, EPICS allows more than one `.` in an EPICS process variable name.

Thanks for uncovering this little gem.  I had not come across it before.  The last common instances of double-dot PVs was excised from the APS control systems many years ago, before ophyd was conceived.

I'm searching for the naming rules in the EPICS documentation.  Last time I asked this topic, I recall the answer was not as clear as I hoped (hoped for a regular expression, got prose instead).

prjemian, 2024-06-19:
@anjohnson : Can you provide a reference here to the rules about naming of EPICS process variables?  If it makes a difference this issue is discussing Channel Access PV names.

anjohnson, 2024-06-19:
CA and PVA servers have their own different limitations and rules for PV names (also called channel names, the terms mean the same thing); the rules for an IOC server are different than for a PCAS or PVA servers. Thus the names you can use depend on the server that publishes the PV, and a generic client application should generally accept and send any string that the user enters (unless it only supports talking to an EPICS motor record say).

A motor record is an object that lives in an IOC, so the characters up to the first `.` specify the record name, and are subject to the limitations described [here](https://docs.epics-controls.org/en/latest/appdevguide/databaseDefinition.html#definitions-8). A `.` character is optional at the end of the PV name, in which case the channel will connect to the record's `VAL` field.

When pointing to a specific field of an IOC record the `.` is required as a separator and followed by the field name; field names have to follow the same rules as a C-99 identifier, see [here](https://docs.epics-controls.org/en/latest/appdevguide/databaseDefinition.html#definitions-1) for more information.

When connecting to an IOC running EPICS 3.15 or later, the field name (or the `.` which is required in this case) may be followed by a field modifier, see [here](https://epics.anl.gov/base/R7-0/8-docs/filters.html) for the modifiers supported by EPICS 7.0.8.

A JSON field modifier may include `.` characters, although only when inside a JSON value (string or number). Thus the PV name that @gazzar is trying to connect to above couldn't belong to an EPICS motor record, but it might refer to a legal channel on some other CA or PVA server.

HTH!

prjemian, 2024-06-19:
@anjohnson Thanks!  That helps me.  We're only talking here about CA (not PVA).

From the first link, notice that a "record instance" does not include the `.` (dot) character.  (Note also that `{` and `}` are absent from this list.)  A "PV name" can include both the record instance name and `.`+field name.  When the latter is not provided, the CA server defaults to `.VAL`.

prjemian, 2024-06-19:
@anjohnson - I notice there is no advice about "field names". (Perhaps that advice is [here](https://docs.epics-controls.org/en/latest/appdevguide/databaseDefinition.html#definitions-1)?)  I assume that such names do not have `.` is a convention?  I recall some IOC software that preceded the EPICS areaDetector provided fields with embedded `.` characters.

anjohnson, 2024-06-19:
Field names are set by the record type. I wrote
> field names have to follow the same rules as a C-99 identifier, see [here](https://docs.epics-controls.org/en/latest/appdevguide/databaseDefinition.html#definitions-1) for more information.

The standard rules for C identifiers can be found [here](https://en.cppreference.com/w/c/language/identifier), but apparently I should have said C-89 instead of C-99 because even that version allowed unicode characters to be used in identifiers (who knew?). On that page you have to set the "Standard revision" pull-down at the top to C-89 to see our definition as we haven't tried using unicode characters in field names, and even if the Perl code processes them correctly the result will almost certainly not be portable to all our target compilers.

anjohnson, 2024-06-19:
Note that I'm reserving the use of subsequent `.` characters as field modifiers in case we want to add support for `DBF_STRUCT` field-types and allow substructure access to the IOC in the future. That would be years away though, don't hold your breath…

gazzar, 2024-06-20:
Thanks for picking up the discussion around this @prjemian. @anjohnson is correct that this isn't a motor record but a field of a [table record](https://epics-modules.github.io/optics/tableRecord.html) (specifically the y translation drive field), which is implemented as a pseudomotor. It has turned out that I can safely redefine my Ophyd device as an EpicsSignalRO in this case, because that position is only being read from for now in my code. However I did try the monkeypatching approach, which works and will be useful if I decide to redefine the device as an EpicsMotor; Thanks @ZLLentz for a practical workaround.