bluesky/ophyd #1204: unhelpful ValueError associated with positioner movement

bruceravel, 2024-07-18:
Just now, I accidentally did this:
```
RE(mv(xafs_y, 173.439, xafs_y, 64.63))
```
(Note, I specified the same motor twice.)

The stack trace ended with this:
```
ValueError: Can not compose overlapping cycles
```

Now, I know what a cycler is, but approximately 0% of the users of my beamline will know what a cycler is.  The likelihood that that error message would result in a phone call rather than the user figuring out the problem themself seems VERY high.   An error message actionable hint would be of value.

bruceravel, 2024-07-18:
Perhaps through a try/except around https://github.com/bluesky/bluesky/blob/main/src/bluesky/plan_stubs.py#L291 and be on the lookout for a ValueError...?

tacaswell, 2024-07-19:
I agree and it is probably worth a PR to cycler to include the collided keys in the error message.

I'm going to move this to the bluesky repo because all of the relevant logic is in bluesky.