bluesky/ophyd #1210: Recommended simulation EPICS IOC to test EpicsMotor

mguijarr, 2024-09-18:
Which is the recommended simulation IOC to test EpicsMotor in Ophyd ?

I mean, to have a realistic behaviour, with real timings etc ?

I just gave a try to https://github.com/epics-motor/motorMotorSim . It does not seem to be very compatible,
since I have errors like:

```
epics> CA.Client.Exception...............................................
    Warning: "User specified timeout on IO operation expired"
    Context: "ca_search_and_connect"
    Source File: ../motorUtil.cc line 210
    Current Time: Wed Sep 18 2024 11:39:45.496673122
..................................................................
motorUtil.cc: getChID(IOC:moving.VAL) error: 80
CA.Client.Exception...............................................
    Warning: "User specified timeout on IO operation expired"
    Context: "ca_search_and_connect"
    Source File: ../motorUtil.cc line 210
    Current Time: Wed Sep 18 2024 11:40:45.496969230
..................................................................
motorUtil.cc: getChID(IOC:alldone.VAL) error: 80
CA.Client.Exception...............................................
    Warning: "User specified timeout on IO operation expired"
    Context: "ca_search_and_connect"
    Source File: ../motorUtil.cc line 210
    Current Time: Wed Sep 18 2024 11:41:45.497185065
..................................................................
motorUtil.cc: getChID(IOC:movingDiff.VAL) error: 80
Failed to connect to IOC:moving or IOC:alldone or IOC:movingDiff.
Check prefix matches Db
```

On the client side, asking for `.position` blocks forever (for example).

coretl, 2024-10-29:
I would suggest using https://github.com/epics-containers/ioc-motorsim

Docs on how to run it are part of the framework tutorials: https://epics-containers.github.io/main/tutorials.html

coretl, 2024-10-29:
Although I don't know if the `alldone` and `moving` PVs are present in the demo. It might need `motorUtil.db` loading...

mguijarr, 2024-10-31:
Thanks for the help @coretl 👍🏻 

I will check this asap

jwlodek, 2024-10-31:
The linked motorSim should work, I've used it to test ophyd devices in the past, I think you might just be missing `motorUtil.db` as mentioned above.