bluesky/ophyd #1218: Provided value of trigger_value is ignored

prjemian, 2024-10-29:
The [docstring](https://github.com/bluesky/ophyd/blob/5df3f5694f5c4c0ac049f0e601512f220e85744b/ophyd/device.py#L139-L141) in the `Device` code says a Component maybe marked as the Signal to use from the `.trigger()` method by setting the Component's `trigger_value` kwarg.

When it comes time to actually use that signal, the provided value is ignored and a `1` is sent: https://github.com/bluesky/ophyd/blob/5df3f5694f5c4c0ac049f0e601512f220e85744b/ophyd/device.py#L1460

From https://github.com/aps-8id-dys/bluesky/issues/99

prjemian, 2024-10-29:
We found this yesterday when applying an external trigger signal from our SoftGlue FPGA to an area detector.  Both devices must be triggered simultaneously.  The FPGA needs a trigger value of `"1!"`.  All [current ophyd uses](repo:bluesky/ophyd trigger_value) of the `trigger_value` show `1` in every case.  Must have been assumption that has worked so far.  With this new FPGA code, the assumption breaks.

Time to use the supplied value.  This [commit](https://github.com/aps-8id-dys/bluesky/commit/d38297a01082c7f0c7ae11202c2d916bb9a75742) (for the APS XPCS instrument) implements the change locally.

prjemian, 2024-10-29:
NOTE: In the case of the XPCS FPGA's trigger signal, the Component should also include `string=True`.

tacaswell, 2024-10-29:
It looks like the bug is there from the first implementation in f5d5c46240cd1a2a13d1a00d5481a76ee4cb9337 from 2015 🤯

PR #1219 (2024-10-29): use provided value of trigger_value
- close #1218

Tested identical change locally at APS as part of https://github.com/aps-8id-dys/bluesky/issues/95 and https://github.com/aps-8id-dys/bluesky/issues/99.
closed 2024-10-29 by tacaswell