bluesky/ophyd #1225: NotImplementedError in _dummy_shim.py when pyepics is not installed

thomashopkins32, 2024-12-09:
I am getting a `NotImplementedError` when trying to run a bluesky setup script without `pyepics` installed. I think a more descriptive error message would help when trying to resolve this issue.

Should `pyepics` be included in https://github.com/bluesky/ophyd/blob/902d2994188b1843b3605ec2e9b2d8b94a4abd54/setup.cfg#L20-L26 ?

If not, how can we make this error more descriptive?

Setup Script:
```python3
from ophyd import Device
from ophyd.signal import EpicsSignal, EpicsSignalRO
from ophyd import Component as Cpt
from ophyd.status import SubscriptionStatus
from bluesky.callbacks.tiled_writer import TiledWriter
from bluesky.callbacks.best_effort import BestEffortCallback
from bluesky.run_engine import RunEngine
from bluesky.protocols import Movable

from bluesky.plan_stubs import mv
from bluesky.plans import count, scan, grid_scan
import time as ttime

from tiled.client import from_uri

# Wait a few seconds for IOC to start
ttime.sleep(3)

def dump_doc_to_stdout(name, doc):
    print("========= Emitting Doc =============")
    print(f"{name = }")
    print(f"{doc = }")
    print("============ Done ============")

RE = RunEngine({})
# RE.subscribe(dump_doc_to_stdout)

bec = BestEffortCallback()
RE.subscribe(bec)

tiled_client = from_uri("http://localhost:8000", api_key="TSCSD")

tw = TiledWriter(tiled_client)

RE.subscribe(tw)

class TSCSDChannel(Device, Movable):

    ramp_rate_set = Cpt(EpicsSignal, "RR:SET", name="ramp_rate", kind="config")
    setpoint = Cpt(EpicsSignal, "SP:SET", name="ramp_rate", kind="config")
    ramp_rate = Cpt(EpicsSignal, "RR", name="ramp_rate", kind="hinted")
    value = Cpt(EpicsSignal, "RB", name="value", kind="hinted")
    at_setpoint = Cpt(EpicsSignalRO, "ATSP", name="at_setpoint", kind="omitted")

    def set(self, value):

        def is_at_rest_callback(value, old_value, **kwargs):
            if value == 1 and old_value == 0:
                return True
            return False
        
        status = SubscriptionStatus(self.at_setpoint, run=False, callback=is_at_rest_callback)
        self.setpoint.put(value)
        return status

class TSCSD(Device):
    chan_1 = Cpt(TSCSDChannel, "1:", name="chan_1")
    chan_2 = Cpt(TSCSDChannel, "2:", name="chan_2")
    chan_3 = Cpt(TSCSDChannel, "3:", name="chan_3")
    chan_4 = Cpt(TSCSDChannel, "4:", name="chan_4")

tscsd = TSCSD("DEV:TSCSD1:", name="tscsd")
[... 125 lines cut ...]
   1841     metadata=metadata,
   1842     all_pvs={read_pv, write_pv},
   1843     **kwargs,
   1844 )
   1846 if read_pv == write_pv:
   1847     self._write_pv = self._read_pv

File ~/anaconda3/envs/TSCSD/lib/python3.11/site-packages/ophyd/signal.py:1067, in EpicsSignalBase.__init__(self, read_pv, string, auto_monitor, name, metadata, all_pvs, timeout, write_timeout, connection_timeout, **kwargs)
   1064     if pv not in self._metadata_key_map:
   1065         self._metadata_key_map[pv] = {}
-> 1067 self._read_pv = self.cl.get_pv(
   1068     read_pv,
   1069     auto_monitor=self._auto_monitor,
   1070     connection_callback=self._pv_connected,
   1071     access_callback=self._pv_access_callback,
   1072 )
   1073 self._read_pv._reference_count += 1
   1075 if not self.__any_instantiated:

File ~/anaconda3/envs/TSCSD/lib/python3.11/site-packages/ophyd/_dummy_shim.py:51, in get_pv(*args, **kwargs)
     50 def get_pv(*args, **kwargs):
---> 51     raise NotImplementedError

NotImplementedError: 
```

tacaswell, 2024-12-09:
I think the value passed to `NotImplementedError` in `get_pv` should have a description that says you either need to install caproto or pyepics.

thomashopkins32, 2024-12-10:
@tacaswell Is there ever a case where you would use ophyd without pyepics or caproto that would also raise this exception? If not, then I can make the change. Otherwise, maybe if they are calling `get_pv` from an `EpicsSignalBase` we can raise the exception there instead. If they are using an `EpicsSignalBase` then we can assume they need pyepics installed (I think).

I'm not sure if there is an equivalent for not having caproto installed.

tacaswell, 2024-12-10:
The history here is that in the beginning we thought ophyd could be a cross-control system layer and by replacing the "shim" layer you could swap between underlying distributed control systems (e.g. epics vs tango).  However, when actually trying this we found that ophyd has too many epics-isms in the design for this to be really practical.   

The shim layer is still useful for switching between pyepics and caproto (and if the world gains a third sync CA binding we can add it) and we want to avoid having a hard EPICS dependency because there is some stuff in ophyd (like the status objects or `Signal`, in principal you can also import class definitions and introspect them (so long as you don't try to connect!) without a CA binding, but I'm less sure that is actually useful).

------

For the purposes of this application, pyepics and caproto are interchangeable: both are synchronous (thread based) CA client libraries.  pyepics uses ctypes to expose libca, caproto is a pure-Python re-implementation. 

-----

We should leave the exception where it as is we already funnel all of the channel creation through the `get_pv` function in the shims.

thomashopkins32, 2024-12-10:
@tacaswell Okay makes sense, thanks for the additional context! I'll add a description to the `NotImplementedError` as you suggested.

PR #1226 (2024-12-11): Add messages to dummy control layer NotImplementedError to install either pyepics or caproto
See the issue for discussion.

Is there a static list of supported control layer "shims" I can use for the error message?

Closes #1225
closed 2024-12-11 by tacaswell