bluesky/ophyd #1232: Correct return types to match current bluesky protocols

evalott100, 2025-02-03:
Ophyd still uses older and less specific return types for some of the bluesky protocols.

``` python
class SomeDevice(Device):
    readable_sig = Component(SignalRO, "prefix")

def read_stuff(signal: Readable):
    pass

my_device = SomeDevice("prefix", name="name")

read_stuff(my_device.readable_sig)
```

gives type errors

```
Argument of type "SignalRO" cannot be assigned to parameter "signal" of type "Readable" in function "read_stuff"
  "SignalRO" is incompatible with protocol "Readable"
    "read" is an incompatible type
      Type "() -> dict[str | Unknown, dict[str, float | Unknown]]" cannot be assigned to type "() -> SyncOrAsync[Dict[str, Reading[Unknown]]]"
        Function return type "dict[str | Unknown, dict[str, float | Unknown]]" is incompatible with type "SyncOrAsync[Dict[str, Reading[Unknown]]]"
          Type "dict[str | Unknown, dict[str, float | Unknown]]" cannot be assigned to type "SyncOrAsync[Dict[str, Reading[Unknown]]]"
    "describe" is an incompatible type
      Type "() -> dict[str | Unknown, dict[str, Unknown]]" cannot be assigned to type "() -> SyncOrAsync[Dict[str, DataKey]]"
        Function return type "dict[str | Unknown, dict[str, Unknown]]" is incompatible with type "SyncOrAsync[Dict[str, DataKey]]"
```

Correcting 

https://github.com/bluesky/ophyd/blob/1e5f03752f6a6b36ae5738ebbdf4f7645129b20d/ophyd/device.py#L563

and

https://github.com/bluesky/ophyd/blob/1e5f03752f6a6b36ae5738ebbdf4f7645129b20d/ophyd/device.py#L590

to match

https://github.com/bluesky/bluesky/blob/88c403ffbfb539edfb74240703caca6c228480b7/src/bluesky/protocols.py#L269

and

https://github.com/bluesky/bluesky/blob/88c403ffbfb539edfb74240703caca6c228480b7/src/bluesky/protocols.py#L287

would fix the above.

There may be other places, e.g

https://github.com/bluesky/ophyd/blob/1e5f03752f6a6b36ae5738ebbdf4f7645129b20d/ophyd/flyers.py#L77

and

https://github.com/bluesky/bluesky/blob/88c403ffbfb539edfb74240703caca6c228480b7/src/bluesky/protocols.py#L312

NoxHarmonium, 2025-05-12:
We're getting the same type errors when trying to pass devices/signals from our beamline library into the Bluesky plan stubs like `read` or `mv`.

I wanted to have a look into it and see if I could help. The first idea I had was to create a test suite to check if the Ophyd devices conform to the various Bluesky protocols. Turns out there is already a comprehensive test suite that does exactly that!

It also seems to be passing at the moment:

https://github.com/bluesky/ophyd/actions/runs/14726448528/job/41330113376#step:11:1537

It does use mypy, and we are using pylance so I suppose it might be that pylance is more strict. I'll see if I can determine whether it is a pylance specific thing.

NoxHarmonium, 2025-05-12:
Interestingly enough, pyright seems happy enough with the test suite too.

https://github.com/NoxHarmonium/ophyd/commit/8d46b830924d04ab2b4ca83186b8188661f7ac11

I'll keep investigating.

NoxHarmonium, 2025-05-13:
I've got a very rough branch in my fork that at least fixes the common classes like Device/Signal for Readable and Movable. 

https://github.com/NoxHarmonium/ophyd/pull/1/files

I went down a bit of a rabbit hole because it seems like the current tests are not actually testing anything. For example, there is a test for the `Hinted` protocol at https://github.com/bluesky/ophyd/blob/main/ophyd/tests/test_bluesky_protocols.py#L32 which is passing, but it doesn't seem like Hinted exists in Bluesky anymore.

For modules that don't exist, mypy is just defaulting to `Any`.

So for example, the following in the following test:

```
from bluesky.protocols import Hinted
foo: Hinted = ophyd.Signal(name='test')
```

`Hinted` is converted to `Any`

```
from bluesky.protocols import Hinted
foo: Any = ophyd.Signal(name='test')
```

and anything is assignable to `Any` so the test always passes.
I'm adding the switches `"--follow-imports=silent", "--disallow-any-unimported"` and going through and fixing them up as I find issues. Hopefully I can work on it a bit more tomorrow.

NoxHarmonium, 2025-05-14:
I've opened a draft PR now which I've tested against our beamline library. Let me know if that seems like the right approach:
https://github.com/bluesky/ophyd/pull/1248/files